SICKAG/sick_safetyscanners #1: /sick_safetyscanners/laser_scan publish frequency is very slow (1.6 - 2Hz)

xdaco, 2018-11-09:
Hi, I am testing Microscan3 Pro  (MICS3-CBAZ55IZ1) . I am experiencing very slow laser_scan messages and occasional TCP error code: 0. The average publish frequency is around 1.6Hz to 2 Hz even after setting the publish frequency to 50Hz with Dynamic reconfigure. Can anyone point me some direction to resolve this ?

puck-fzi, 2018-11-09:
Hi, The TCP error code 0 means the tcp connection was setup successful. Any other number then 0 means there are problems. I will change the output to be more clear in upcoming releases.

The publish frequency in the dynamic reconfigure means that every n-th scan will be published. So if you set it to 50 every 50th scan will be published. If you set it to 1, every scan will be published. 

Does this resolve the issue? If not, there might be some settings in the scanner which lead to only a low amount of data being send to the driver.

xdaco, 2018-11-10:
Thank you so much for your reply.  Yes, this clears the issue. TCP error 0 should not be reported as ROS_ERROR as this is misleading.  Also, I think the dynamic reconfigure window is misleading as one can assume that publishing frequency as the message publishing frequency as this is not consistent with the next param viewing angle.

puck-fzi, 2018-11-11:
Glad i could clear it up. I will look into making it clearer.

closed 2018-11-11 by puck-fzi

chadrockey, 2018-12-05:
@xdaco In pull request #2 I pushed this commit:
https://github.com/SICKAG/sick_safetyscanners/pull/2/commits/bc46acdb1018e02c8d58fc26a91787f399fb89e1

It changes the 'publish_frequency' name and behavior to that of 'skip' which has been used commonly in other ROS drivers.  Does this meet the expectations for behavior that you specified here?

http://wiki.ros.org/hokuyo_node#Dynamically_Reconfigurable_Parameters