SICKAG/sick_safetyscanners #4: TCP Connection Mode

chadrockey, 2018-12-05:
Does the microscan3 support the TCP connection mode?  What would it require to use this mode?

I think most of the other ROS lasers use TCP by default so that the host_ip isn't required in the connection information.  This might make it easier for newer users/users with complicated configurations to run the software.

puck-fzi, 2018-12-06:
There are two possibilities to access the scan data from the sensor. 

The here implemented method uses the continuous data stream via UDP data packages. 
It would be possible to poll the latest Scan from the sensor. But this has to be actively triggered by the host each time a scan is required and there is no continuous option for the TCP packages.

Therefore the TCP connections are used in this package to request additional variables from the driver, such as the protective and warning fields.

chadrockey, 2018-12-07:
Yes, I agree that polling is not ideal.

Do you know if the TIMs support TCP continuous data streams?

My motivation is to not require the host_ip or host_port parameters for basic users.  Am I missing the option that allows this behavior?

puck-fzi, 2018-12-11:
I think the TIMs support TCP, but I never worked with them.

I think the there is no good option to skip the host IP. The only thing i could think of as minor improvement is to read the configured host_ip from the sensor. But that probably only helps in a few cases, sicne as soon as you want to connect with a different host you have to change the parameter again.