SICKAG/sick_safetyscanners #5: How To Determine Port Number

Jmeyer1292, 2019-01-18:
Greetings. I have a Microscan3 Pro I'm trying to tie into a computer application and I'm having a little trouble. I have:

1. Built the software
2. Verified that the sensor is operational (attached to a PLC, shows green)
3. Verified that the sensor and computer are on the same network (I can ping it).
4. Enabled data output in the Sick configuration tool.

With regards to data output, I set:
```
Send Mode = On request
Selection data content = All boxes checked
```
The driver launches and shows no visible errors but information is not flowing: If I echo the sensor's scan topic, nothing shows up. I can call the service and information _does_ show up. So it must be something in the ROS or scanner configuration.

How do I tell what TCP port I should be setting in the `roslaunch file`? Does the scanner need to be configured to stream to the target continuously?

Thanks,
Jon

puck-fzi, 2019-01-19:
Hello Jon,

if you enable the continuous streaming you get the desired information? Did I grasp you addendum correct?

The TCP port normally defaults to 2122, unless you have explicitly changed that in your sensor configuration it should be the same. This appears to be working correctly if the services produce output, since they use the TCP connection. For the scans they solely use the UDP connection. Therefore please ensure that the Host_IP is set correctly and all data is enabled and continuously streamed.

Jmeyer1292, 2019-01-19:
Hey, thank you for the response. My addendum refers to the `send mode` set in SICK's configuration tool. Specifically, I set send mode to `on request and continuously to IP/PORT`. In this scenario I see the data show up. Is this the way the sensor is meant to be used? Or should the TCP control protocol start/stop the streaming? I'll dig into it more on Monday. I'm sure I'm missing something obvious.

Thanks again.

puck-fzi, 2019-01-22:
Hi Jon, 
Yes please enable the continuous streaming for the data to be received. Otherwise you can only poll the data from the sensor which is not implemented since the continuous streaming is more convenient and all scans are published that way. The TCP connection is then used to specify which Host the data should be sent to. You can further enable or disable a channel of the data. At the moment the sensor only supports one channel, but later there should be up to four channels which can be individually configured to serve different receivers.

Hope that clarifies your questions.

closed 2019-01-28 by puck-fzi