SICKAG/sick_safetyscanners #11: Scan topic has no new messagesof master: 220fc4419fc9ad50acfae0ebdf58fbde9fec95eb) (At the tip

wongkaiweng, 2019-04-24:
Hi, 

I am currently using a micoScan3 Pro and I have no problem getting incoming scan messages with 3251beac6e0662c25f15c4aa19a2e0c8c20482a6.  However, I just pulled the latest commits on master (220fc4419fc9ad50acfae0ebdf58fbde9fec95eb) and the scan topic is no longer publishing any messages. Below is the output when I ran `roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=192.168.1.10 host_ip:=192.168.1.9`:

```
started core service [/rosout]
process[sick_safetyscanners/sick_safetyscanners-2]: started with pid [24022]
[ WARN] [1556137453.117857792]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO] [1556137453.121330700]: Starting SickSafetyscanners
[ INFO] [1556137453.121566843]: UDP client is setup
[ INFO] [1556137453.121653550]: Started SickSafetyscanners
[ INFO] [1556137453.121783144]: Enter io thread
[ INFO] [1556137453.121928423]: TCP client is setup
[ INFO] [1556137453.122795406]: TCP connection successfully established.
[ INFO] [1556137453.180060620]: Successfully opened Cola2 session with sessionID: 1030567907
[ INFO] [1556137453.214923944]: Command Variable Acknowledged.
[ INFO] [1556137453.254917309]: Successfully closed Cola2 session with sessionID: 1030567907
[ INFO] [1556137453.255163083]: TCP Connection successfully shutdown
[ INFO] [1556137453.255273480]: TCP Socket successfully closed.
[ INFO] [1556137453.255371477]: TCP client is setup
[ INFO] [1556137453.256297514]: TCP connection successfully established.
[ INFO] [1556137453.310115135]: Successfully opened Cola2 session with sessionID: 3334719619
[ INFO] [1556137453.360088150]: Command Method Acknowledged.
[ INFO] [1556137453.415511354]: Successfully closed Cola2 session with sessionID: 3334719619
[ INFO] [1556137453.415788200]: TCP Connection successfully shutdown
[ INFO] [1556137453.415900401]: TCP Socket successfully closed.
[ INFO] [1556137453.415949513]: Successfully launched node.
```
I am not seeing any errors so I'm not sure why that is the case.

puck-fzi, 2019-04-25:
Hi, I am out of office for another two weeks. I will check afterwards if I can find an issue. If you find anything which causes the troubles please let me know. Cheers

Tent-Beaumont, 2019-04-26:
Hi Catherine, this is unfortunately nothing we at Sick can fix. If using the previous version is really not an option, please contact me and we will try to figure out a faster solution.

Sprinzzzz, 2019-05-06:
@wongkaiweng Did you find a solution to this issue? I am facing the same issue. The rostopic echo /sick_safetyscanners /laserscan is empty.

puck-fzi, 2019-05-06:
Hi, it appears there was a regression on the latest commits. Please use the commit 3251bea for the time being and I will fix the issue as soon as possible.

puck-fzi, 2019-05-06:
Hi, please check if the latest commit resolved this issue for you. On my side the error is gone and the correct angles are used on startup.

closed 2019-05-06 by puck-fzi