SICKAG/sick_safetyscanners #12: Issue in viewing the scan messages

Sprinzzzz, 2019-05-06:
Greetings. 
I have a Microscan3 Pro I'm trying to view the data in ROS and I'm having a little trouble. I have:

Built the software
Verified that the sensor is operational (attached to a PLC, shows green)
Verified that the sensor and computer are on the same network (I can ping it).
Enabled data output in the Sick configuration tool.
With regards to data output, I set:

```
Send Mode = On request and additionally to target computer
Selection data content = All boxes checked
IP- 192.168.1.10
UPD-6060
Router- 192.168.1.254
send-every measuring
```
My IP details of the PC and router are below,

[image]

The commanad I used

`roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=192.168.1.93 host_ip:=192.168.1.10 host_udp_port:=6060`

The sensor IP in the sick configuration is 192.168.1.93 with subnet-255.255.255.0 and router-192.168.1.254

The driver launches and shows no visible errors but information is not flowing: If I echo the sensor's scan topic, nothing shows up. I can call the service and information does show up. 

How can I view the data of the scan messages? What am I doing wrong?

Thanks,
Sprinzzz

Note: This is similar to the issue that was closed a month ago but a bit different as it doesn't work even with Send Mode = On request and additionally to target computer.

puck-fzi, 2019-05-06:
Hi Sprinzzz, it appears that there was a regression on the last commit. I am looking into it as of now. There will be an update soon. Please use the previous commit 3251bea for the moment. I will update the repo as soon as i have a fix. Please excuse the inconvinience

puck-fzi, 2019-05-06:
Hi, please check if the latest commit resolved this issue for you. On my side the error is gone and the correct angles are used on startup.

closed 2019-05-06 by puck-fzi