SICKAG/sick_safetyscanners #13: How can I view 2 laser scan data?

Sprinzzzz, 2019-05-06:
Hello,

I am using 2 microscan3 pro for my project. Could you please let me know if there is a way to view 2 scanners data with this package? I tried it but I could view the active topics only one scanner and similarly one scanner data in rviz (tried to create 2 laser scanners in rviz).

Thanks in advance for your suggestion.

Best regards
Sprinzzz

puck-fzi, 2019-05-06:
Hi Sprinzzz,
for that you could start two instances of the driver in different namespaces, one for each scanner.
If you have a urdf with the setup of the scanners and the frame_id of the sensor is correctly set you should get the data in the corresponding frames ad can display them both.

Sprinzzzz, 2019-05-07:
That works. Thank you.

closed 2019-05-08 by puck-fzi