SICKAG/sick_safetyscanners #14: How to view protective fields

Aradan2, 2019-05-07:
Hello,

I am currently using a microscan3 pro. I am able to get data from the scanner and display it using rviz. It looks like this:
[image]
I am now wondering, if it's possible, to also get protective fields, that are previously defined in the scanner via sick-safetydesigner, and show them together with the measurement-data i already get, in one rviz display.

Thanks in advance for any suggestion.

Best regards

puck-fzi, 2019-05-07:
Hi, this is not directly possible in rviz. Since the protective fields are returned by a service call (~/field_data) since they are not continuously streamed. You will get the field dimensions and the monitoring cases they belong to. Together with the extended laserscan message you can deduct the currently active fields and if they are breached. You would need to write your own visualization at the moment. The driver will give you all the necessary information. You would get the field data once at the beginning and then subscribe to the extended laser scan.

closed 2019-05-07 by puck-fzi

Aradan2, 2019-05-09:
Thank you for your advice!
Is there also a way to get the refercence contour fields?
In the FieldData.h i only can find protective fields and warning fields.
And the FieldData.srv uses FieldMsg, which only contains a bool whether it is a protective field or not...
So what would i have to call, to get information about the reference contour fields?
Sorry if this is a stupid question, but I'm an absolutly newbie to ROS.

melecmat, 2025-09-18:
Hello, I am trying to visualize the contour fields in ROS and stumbled upon this ancient issue. Has there been any updates? 
I was looking into the `requestFieldDataInColaSession` method. I uncommented the part that skips invalid fields. 
Looking at index 0, I then have an invalid field with no data (and eval_method=127) - I was expecting to find there the contour data? Are the contour data now saved somewhere else, or is the driver just unable to read them?
Additionally I find it weird that the monitoring cases report that the field number of the contour field should be 82. When I visualize the field 82 it does not look like the contour I am expecting at all.