SICKAG/sick_safetyscanners #15: UDP data is streaming, but no messages

mikepurvis, 2019-05-10:
I have continuous data on, and UDP messages arriving from my two devices:

```
05:00:29.096007 IP 10.255.255.21.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.096570 IP 10.255.255.22.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.101142 IP 10.255.255.21.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.101584 IP 10.255.255.22.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.105383 IP 10.255.255.21.6000 > 10.255.255.1.6000: UDP, length 512
05:00:29.105900 IP 10.255.255.22.6000 > 10.255.255.1.6000: UDP, length 512
05:00:29.136018 IP 10.255.255.21.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.136673 IP 10.255.255.22.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.141156 IP 10.255.255.21.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.141592 IP 10.255.255.22.6000 > 10.255.255.1.6000: UDP, length 1460
05:00:29.145381 IP 10.255.255.21.6000 > 10.255.255.1.6000: UDP, length 512
05:00:29.145896 IP 10.255.255.22.6000 > 10.255.255.1.6000: UDP, length 512
```

I'm launching the node like so:

```
$ roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=1
0.255.255.21 host_ip:=10.255.255.1 host_udp_port:=6000 --screen

started roslaunch server http://localhost:42177/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.14.3
 * /sick_safetyscanners/sick_safetyscanners/angle_end: 0.0
 * /sick_safetyscanners/sick_safetyscanners/angle_start: 0.0
 * /sick_safetyscanners/sick_safetyscanners/application_io_data: True
 * /sick_safetyscanners/sick_safetyscanners/channel_enabled: True
 * /sick_safetyscanners/sick_safetyscanners/derived_settings: True
 * /sick_safetyscanners/sick_safetyscanners/frame_id: scan
 * /sick_safetyscanners/sick_safetyscanners/general_system_state: True
 * /sick_safetyscanners/sick_safetyscanners/host_ip: 10.255.255.1
 * /sick_safetyscanners/sick_safetyscanners/host_udp_port: 6000
 * /sick_safetyscanners/sick_safetyscanners/intrusion_data: True
 * /sick_safetyscanners/sick_safetyscanners/measurement_data: True
 * /sick_safetyscanners/sick_safetyscanners/sensor_ip: 10.255.255.21
 * /sick_safetyscanners/sick_safetyscanners/skip: 0
 * /sick_safetyscanners/sick_safetyscanners/time_offset: 0.0
 * /sick_safetyscanners/sick_safetyscanners/use_persistent_config: False

NODES
  /sick_safetyscanners/
    sick_safetyscanners (sick_safetyscanners/sick_safetyscanners_node)

auto-starting new master
process[master]: started with pid [4025]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f86ceee2-73a6-11e9-b61d-2046a1034943
process[rosout-1]: started with pid [4036]
started core service [/rosout]
process[sick_safetyscanners/sick_safetyscanners-2]: started with pid [4040]
[ WARN] [/sick_safetyscanners/sick_safetyscanners] [1557549660.400232841]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.406163406]: Starting SickSafetyscanners
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.406452354]: UDP client is setup
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.406551215]: Started SickSafetyscanners
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.406668071]: Enter io thread
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.406704012]: Reading Type code settings
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.406820722]: TCP client is setup
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.407135105]: TCP connection successfully established.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.450311899]: Successfully opened Cola2 session with sessionID: 787905971
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.485515449]: Command Variable Acknowledged.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.525477276]: Successfully closed Cola2 session with sessionID: 787905971
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.525600481]: TCP Connection successfully shutdown
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.525665106]: TCP Socket successfully closed.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.525720155]: TCP client is setup
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.526234171]: TCP connection successfully established.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.585853851]: Successfully opened Cola2 session with sessionID: 2281629719
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.620753825]: Command Method Acknowledged.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.660672609]: Successfully closed Cola2 session with sessionID: 2281629719
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.660869332]: TCP Connection successfully shutdown
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.661073188]: TCP Socket successfully closed.
[ INFO] [/sick_safetyscanners/sick_safetyscanners] [1557549660.661134776]: Successfully launched node.
```

But I get nothing on the `scan` or `raw_data` topics. When I put a breakpoint in the `SickSafetyscanners::processUDPPacket` function, it never fires, suggesting that something about config is not resulting the socket being set up properly.

puck-fzi, 2019-05-13:
Hello mikepurvis, 
with the same launch parameters I can start the driver without trouble here. So the processUDP is called here. 
The datastream is still incoming when the driver is launched? Or does the datastream stop once the configuration is sent to the sensor?

mikepurvis, 2019-05-13:
I ran `tcpdump` while the driver was running. Would it be helpful to get a full pcap? Are there particular settings I should check through Safety Designer?

puck-fzi, 2019-05-14:
As of now, I have not had a configuration were I did not get any data. There might be something i missed though.

Could you check if the function void AsyncUDPClient::handleReceive in sick_safetyscanners/src/communication/AsyncUDPClient.cpp is triggered. This should be the very first function processing the UDP Packets. But it should show a ROS_ERROR when it fails. And for the socket, if it does not come up properly it should throw an exception and show an error. Therefore I can not really pin point where the problem might reside.

Do you have any special setup? I tested it here with a safetyscanner directly connected to a computer and with a flexisoft in between, both cases do work for me.

You might want try to set the skip parameter to one maybe. With a Long Range Safetyscanner I had at the troubles that when I was accessing variables with the Cola Protocol and had the continuous data stream on, the sensor sometimes stopped sending data. But this should be visible in the datadump.

Furthermore the last commit as referenced in issue #11 there were troubles with the start and end angle. You might want to try setting both to different values so they are not equal. It should be fixed in the latest commit and should not happen in the commit  3251bea. 

Since I cannot reproduce the error I am mostly guessing if you have anything special configured or setup please let me know so I can test it on my end.

closed 2019-06-03 by puck-fzi

mikepurvis, 2019-06-03:
Sorry, yes, it turned out that ufw was blocking inboudn udp messages. This wasn't visible to tcpdump running as root, but the driver never saw them. Thanks for the help!

puck-fzi, 2019-06-03:
Thank you for the clarification