SICKAG/sick_safetyscanners #16: How can I suscribe to a message that tells me if there is an object in a warning/protection zone?

derektan95, 2019-06-03:
Hi, 

I am new to this LIDAR and would like to seek some help with this. After uploading a new set of warning and protection zones to the LIDAR via the Sick software provided, I am trying to receive boolean values that can indicate if there is an object within each of the zones in the different Monitoring planes. Wonder if there is any specific messages that I can subscribe to to achieve this?

puck-fzi, 2019-06-03:
Hi derektan95, 
the topic ( ~/output_paths) gives the current status of each output path in the current monitoring case. If you want to get the field dimensions you can get this via service call.

closed 2019-06-03 by puck-fzi

derektan95, 2019-06-03:
Hi puck-fzi, 

I tried subscribing to the output path message but there seem to be different combinations of true and false output (for the first 4 boolean values). For the same monitoring case, if I only set 1 

==Status message==
Protective Zone: [False, False, False, False ... ]
Warning Zone: [True, False, False, False ... ]
No obstruction: [True, True, False, False...]

However, these values change as I add more Fields or Field Sets. May I know how I can figure out what combination of boolean values would imply protection/warning/no obstruction cases?

derektan95, 2019-06-03:
Hi puck-fzi,

I tried subscribing to the output path message but there seem to be different combinations of true and false output (For the same monitoring case):

==Status message==
Protective Zone: [False, False, False, False ... ]
Warning Zone: [True, False, False, False ... ]
No obstruction: [True, True, False, False...]

However, these values change as I add more Fields or Field Sets. May I know how I can figure out what combination of boolean values would imply protection/warning/no obstruction cases?