SICKAG/sick_safetyscanners #17: Suscribing to ~/output_paths

derektan95, 2019-06-03:
Hi puck-fzi,

I saw that you closed the previous thread so I will post another question here instead. I tried subscribing to the output path message but there seem to be different combinations of true and false output within the "Status" message (For the same monitoring case):

Protective Zone: [False, False, False, False ... ]
Warning Zone: [True, False, False, False ... ]
No obstruction: [True, True, False, False...]

However, these combination of values corresponding to protection/warning/no obstruction case changes as I add more Fields or Field Sets. May I know how I can figure out what combination of boolean values would imply protection/warning/no obstruction cases?

puck-fzi, 2019-06-03:
Hi the topic returns the following values:
bool[] status
bool[] is_safe
bool[] is_valid
int32 active_monitoring_case

the status tells you if the field is breached, the is_safe, if it is a safety field oder protective, and is valid if the field is configured. the active monitoring case tells you which monitoring case is currently active.

The service call ~/field_data gives you all fields and their ranges and returns an array of the monitoring cases (sick_safetyscanners/MonitoringCaseMsg[] monitoring_cases) there you can see which fields belong to which monitoring case

derektan95, 2019-06-03:
Hi, 

Thanks for explaining the meaning for each of the return values from the output message. To explain my question further, I tried displaying the output_paths message while gradually placing an object from outside the warning zone to inside the warning zone, and the subsequently to within the protection zone. It seems like only the first few values of the boolean array "Status" has changed, while the other arrays (is_safe, is_valid) remained the same. 

It seems like different combination of boolean values within the "Status" array corresponds to whether an object is outside or inside the different zones. Is there a way to understand the what combination of boolean values would correspond to which state (within the same monitoring case)? Or must we do this experimentally to see what the combination of true/false values would lead to each state (and thereafter hardcoding them)?

puck-fzi, 2019-06-03:
Hi, yes the status describes if the field is breached. The data is directly inferred from the data from the sensor, so if you configured the fields that the first is the largest e.g the warning field and the second is the protective field. Then the first element of each array belongs to the warning field, and the second to the protective. It should be according to the visuals on the sensor itself

puck-fzi, 2019-06-03:
Meaning each field is configured on its own... The zones you are describing are just a combination of different fields as you placed them in the safety designer. So yes your zones will be a combination from the different status values

closed 2019-06-04 by puck-fzi

derektan95, 2019-06-04:
Hi, 

Thank you so much for your clarification. I finally got this to work. Just a side question: 

Earlier on, I defined 4 fields(mixture of warning and protection zones within monitoring case 1). I experimented with setting my  "Safe Output" channels on the Safety Designer software (Under Monitoring cases) to be channels 3, 4, 6, 7 instead of 1, 2, 3, 4. Is there a way to obtain which Safe Output channels is utilized by the LiDAR from any of the messages/services? I tried extracting information from the "monitoring_cases_in[0].fields_valid" from the service message but it would only show "1 1 1 1 0 0 0 0" instead of the intended "0 0 1 1 0 1 1 0". Trying "monitoring_cases_in[0].fields" does not demonstrate which channels is utilized too. Hoping to hear from you soon, thanks!