SICKAG/sick_safetyscanners #18: Safe Output channels

derektan95, 2019-06-04:
Hi,

Thank you so much for your clarification about the previous issue. I finally got this to work. Just a side question:

Earlier on, I defined 4 fields(mixture of warning and protection zones within monitoring case 1). I experimented with setting my "Safe Output" channels on the Safety Designer software (Under Monitoring cases) to be channels 3, 4, 6, 7 instead of 1, 2, 3, 4. Is there a way to obtain which Safe Output channels is utilized by the LiDAR from any of the messages/services? I tried extracting information from the monitoring_cases_in[0].fields_valid" from the service message but it would only show "1 1 1 1 0 0 0 0" instead of the intended "0 0 1 1 0 1 1 0". Trying "monitoring_cases_in[0].fields" does not demonstrate which channels is utilized too. Hoping to hear from you soon, thanks!

puck-fzi, 2019-06-04:
As far as I know, this is not possible. The values are send via cola2 protocol. I will have a look if there is any specification which can resolve that. But just from the indices of the fields this is not possible.

derektan95, 2019-06-04:
Ok, thank you!

puck-fzi, 2019-06-04:
Since you only can have 8 fields active at the current state. This will not show in the output paths message. With the service call you can get all fields...But i think they are ordered by indices.

derektan95, 2019-06-04:
Yes, unfortunately the output from the service call will always begin with the channels. It would 1,2,3,4,0,0,0,0 or 1,3,6,7,0,0,0,0, but never 1,0,2,0,0,3,4,0

puck-fzi, 2019-06-04:
Okay, now I think I fully understood. Yes you will always get the ids of the field belonging to the monitoring case... si it will be always the first fields. This is the data as sent by the scanner via cola2 protocol. So to the best of my knowledge it is not possible.

derektan95, 2019-06-04:
I see. Thanks for letting me know! Your information have been really helpful

closed 2019-06-05 by puck-fzi