SICKAG/sick_safetyscanners #19: Memory leak in sick_safetyscanners_node

stefan-weber-innok, 2019-07-11:
Hi,

We are using this driver with an outdoorScan3 and a microScan3. In both cases we are experiencing a memory leak in the official melodic version of sick_safetyscanners (1.0.2) under Ubuntu 18.04.
[image]
As can be seen after 10 hours of runtime, the node is at close to 4GB of ram usage and it only gets worse.
We did not investigate any further than this.

If we can provide any further information we are happy to do so.

puck-fzi, 2019-07-15:
Hi, thanks for letting me know.
I will investigate where this might occur.

puck-fzi, 2019-07-15:
Hi, I found a memory leak due to not correctly erasing data.
With a short test on my site the memory consumption does not appear to be growing anymore.
Please check out the latest commit in this repository. A release will be triggered soon again.

closed 2019-07-15 by puck-fzi

puck-fzi, 2019-07-18:
A version 1.0.3 is already in the ros shadow repository. The next sync will include the package on the main repository. The syncs are manually made. So it will take around 2-3 weeks until the new release is in the official apt package.