SICKAG/sick_safetyscanners #20: 90° offset in scan

maracuya-robotics, 2019-07-22:
While `start_angle` and `end_angle` work as expected (`0.0` is in x direction), the actual scan is rotated by 90°.

puck-fzi, 2019-07-22:
Hi,

could you elaborate what exactly you mean? The 0.0 is supposed to be in the x direction. But i don't understand what you mean with the actual scan is rotated?

maracuya-robotics, 2019-07-22:
Sorry, I think this was already fixed in 1.0.3 (I'm still on 1.0.2 but used the documentation of 1.0.3).

I meant the -90° "Sick offset".

closed 2019-07-22 by maracuya-robotics

puck-fzi, 2019-07-22:
Alright. For the 1.0.3 version you currently have to check out this repository. The debian package will be available soon. It still awaits the next sync cycle for melodic and kinetic, which should be triggered in the next weeks. The announcements will be made here: https://discourse.ros.org/c/release