SICKAG/sick_safetyscanners #22: ROS2 Driver?

maxlein, 2019-09-24:
Any info if there will be a ROS2 driver or a timeline for releasing one?

puck-fzi, 2019-09-24:
At the moment there is no ROS2 Driver planned, but there will be a release soon of a standalone cpp driver, which is currently under development. Using that it would be possible to define access to the data using different frameworks.

maxlein, 2019-09-24:
But it should also be possible to take this driver here and replace the ROS Wrapper class, shouldn't it?

JinTongxing, 2019-09-24:
It seems that there is something wrong with TCP, and I couldn't get any infomation from the topics after launching, could u please give me any advice? The following is the log file.
[ INFO] [1569329672.065299785]: UDP client is setup
[ INFO] [1569329672.065324163]: Started SickSafetyscanners
[ INFO] [1569329672.065368485]: Reading Type code settings
[ INFO] [1569329672.065386480]: Enter io thread
[ INFO] [1569329672.065477928]: TCP client is setup
[ INFO] [1569329672.066099410]: TCP connection successfully established.
[ INFO] [1569329672.101292362]: Successfully opened Cola2 session with sessionID: 3778673419
[ INFO] [1569329672.136279847]: Command Variable Acknowledged.
[ INFO] [1569329672.181238439]: Successfully closed Cola2 session with sessionID: 3778673419
[ INFO] [1569329672.181627093]: TCP Connection successfully shutdown
[ INFO] [1569329672.181812769]: TCP Socket successfully closed.
[ INFO] [1569329672.182133263]: TCP client is setup
[ INFO] [1569329672.183164115]: TCP connection successfully established.
[ INFO] [1569329672.231247679]: Successfully opened Cola2 session with sessionID: 4100953995
[ WARN] [1569329672.261071563]: Command Method Not Accepted.
[ INFO] [1569329672.300958709]: Successfully closed Cola2 session with sessionID: 4100953995
[ INFO] [1569329672.301298883]: TCP Connection successfully shutdown
[ INFO] [1569329672.301438799]: TCP Socket successfully closed.
[ INFO] [1569329672.301593240]: Successfully launched node.

puck-fzi, 2019-09-24:
@maxlein Yes this would be possible as well, this is the similar approach of the driver which is realeased soon. Beware that there is some ROS logging in other classes...That is the only breach of of ROS inside the algorithm, you can write a Wrapper for the logging as well.

The main algorithms will stay the same in both releases, so yes feel free to change ROS1 to ROS2 in the Wrapper class.

closed 2019-09-24 by puck-fzi

SomaGallai, 2020-06-09:
Hey @maxlein have you created the wrapper for ROS2, I have been struggling with this lately as I am new to ROS, I could use some pointers if you could help.