SICKAG/sick_safetyscanners #23: read data from MicroScan3 Core

JinTongxing, 2019-09-24:
It seems that there is something wrong with TCP, and I couldn't get any infomation from the topics after launching, could u please give me any advice? The following is the log file.
[ INFO] [1569329672.065299785]: UDP client is setup
[ INFO] [1569329672.065324163]: Started SickSafetyscanners
[ INFO] [1569329672.065368485]: Reading Type code settings
[ INFO] [1569329672.065386480]: Enter io thread
[ INFO] [1569329672.065477928]: TCP client is setup
[ INFO] [1569329672.066099410]: TCP connection successfully established.
[ INFO] [1569329672.101292362]: Successfully opened Cola2 session with sessionID: 3778673419
[ INFO] [1569329672.136279847]: Command Variable Acknowledged.
[ INFO] [1569329672.181238439]: Successfully closed Cola2 session with sessionID: 3778673419
[ INFO] [1569329672.181627093]: TCP Connection successfully shutdown
[ INFO] [1569329672.181812769]: TCP Socket successfully closed.
[ INFO] [1569329672.182133263]: TCP client is setup
[ INFO] [1569329672.183164115]: TCP connection successfully established.
[ INFO] [1569329672.231247679]: Successfully opened Cola2 session with sessionID: 4100953995
[ WARN] [1569329672.261071563]: Command Method Not Accepted.
[ INFO] [1569329672.300958709]: Successfully closed Cola2 session with sessionID: 4100953995
[ INFO] [1569329672.301298883]: TCP Connection successfully shutdown
[ INFO] [1569329672.301438799]: TCP Socket successfully closed.
[ INFO] [1569329672.301593240]: Successfully launched node.

puck-fzi, 2019-09-24:
and you have not setup any other parameters?

JinTongxing, 2019-09-24:
> Hi, to be able to check what the problem is I would need more information, as in what parameters did you set?
> You get the warning: [ WARN] [1569329672.261071563]: Command Method Not Accepted.
> Therefore it seems that some setup in the communication settings are off. So could you give me details on what are you passing to the sensor.

When I run it in a terminal rather than IDE, the output is like:
[ INFO] [1569332478.139647220]: Starting SickSafetyscanners
[ INFO] [1569332478.139699491]: UDP client is setup
[ INFO] [1569332478.139714614]: Started SickSafetyscanners
[ INFO] [1569332478.139753209]: Reading Type code settings
[ INFO] [1569332478.139768755]: Enter io thread
[ INFO] [1569332478.139787376]: TCP client is setup
[ERROR] [1569332481.218416367]: TCP error code: 113
[ERROR] [1569332481.218968296]: Error in tcp handle send and receive: 32

JinTongxing, 2019-09-24:
> Hi, to be able to check what the problem is I would need more information, as in what parameters did you set?
> You get the warning: [ WARN] [1569329672.261071563]: Command Method Not Accepted. [...]

I want run "roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=192.168.1.2 host_ip:=192.168.1.249“, the 192.168.1.249 is the ip of my computer

puck-fzi, 2019-09-24:
Okay TCP Error code 113 means "No route to host", can you ping your sensor from the system?

JinTongxing, 2019-09-24:
> Okay TCP Error code 113 means "No route to host", can you ping your sensor from the system?

I'm sorry, I made a mistake just now.. After good connection and ping, the log infomation is still like:
jintongxing@jintongxing-Inspiron-5488:~/sick_lidar$ roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=192.168.1.2 host_ip:=192.168.1.249
... logging to /home/jintongxing/.ros/log/a1fb4040-dea9-11e9-93b0-00e04c680109/roslaunch-jintongxing-Inspiron-5488-13743.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jintongxing-Inspiron-5488:34753/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_safetyscanners/sick_safetyscanners/angle_end: 0.0
 * /sick_safetyscanners/sick_safetyscanners/angle_start: 0.0
 * /sick_safetyscanners/sick_safetyscanners/application_io_data: True
 * /sick_safetyscanners/sick_safetyscanners/channel_enabled: True
 * /sick_safetyscanners/sick_safetyscanners/derived_settings: True
 * /sick_safetyscanners/sick_safetyscanners/frame_id: scan
 * /sick_safetyscanners/sick_safetyscanners/general_system_state: True
 * /sick_safetyscanners/sick_safetyscanners/host_ip: 192.168.1.249
 * /sick_safetyscanners/sick_safetyscanners/host_udp_port: 0
 * /sick_safetyscanners/sick_safetyscanners/intrusion_data: True
 * /sick_safetyscanners/sick_safetyscanners/measurement_data: True
 * /sick_safetyscanners/sick_safetyscanners/sensor_ip: 192.168.1.2
 * /sick_safetyscanners/sick_safetyscanners/skip: 0
 * /sick_safetyscanners/sick_safetyscanners/time_offset: 0.0
 * /sick_safetyscanners/sick_safetyscanners/use_persistent_config: False

NODES
  /sick_safetyscanners/
    sick_safetyscanners (sick_safetyscanners/sick_safetyscanners_node)

ROS_MASTER_URI=http://localhost:11311

process[sick_safetyscanners/sick_safetyscanners-1]: started with pid [13760]
[ WARN] [1569332860.185127881]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO] [1569332860.191051314]: Starting SickSafetyscanners
[ INFO] [1569332860.191329593]: UDP client is setup
[ INFO] [1569332860.191444736]: Started SickSafetyscanners
[ INFO] [1569332860.191588871]: Enter io thread
[ INFO] [1569332860.191674740]: Reading Type code settings
[ INFO] [1569332860.191836794]: TCP client is setup
[ INFO] [1569332860.192974189]: TCP connection successfully established.
[ INFO] [1569332860.227126292]: Successfully opened Cola2 session with sessionID: 4061413135
[ INFO] [1569332860.256936570]: Command Variable Acknowledged.
[ INFO] [1569332860.257110856]: Type Code: MICS3-ABAZ55IZ1
[ INFO] [1569332860.302182952]: Successfully closed Cola2 session with sessionID: 4061413135
[ INFO] [1569332860.302594908]: TCP Connection successfully shutdown
[ INFO] [1569332860.302771176]: TCP Socket successfully closed.
[ INFO] [1569332860.302979459]: TCP client is setup
[ INFO] [1569332860.304183019]: TCP connection successfully established.
[ INFO] [1569332860.357287796]: Successfully opened Cola2 session with sessionID: 2772603955
[ WARN] [1569332860.387021298]: Command Method Not Accepted.
[ INFO] [1569332860.426882616]: Successfully closed Cola2 session with sessionID: 2772603955
[ INFO] [1569332860.427215350]: TCP Connection successfully shutdown
[ INFO] [1569332860.427370655]: TCP Socket successfully closed.
[ INFO] [1569332860.427464032]: Successfully launched node.

JinTongxing, 2019-09-24:
> ok, then please try to build the driver from source and let me know if that helps

sorry, the problem still exists with the same log information:
jintongxing@jintongxing-Inspiron-5488:~/catkin_ws$ roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=192.168.1.2 host_ip:=192.168.1.249
... logging to /home/jintongxing/.ros/log/a1fb4040-dea9-11e9-93b0-00e04c680109/roslaunch-jintongxing-Inspiron-5488-16672.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jintongxing-Inspiron-5488:37491/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_safetyscanners/sick_safetyscanners/angle_end: 0.0
 * /sick_safetyscanners/sick_safetyscanners/angle_start: 0.0
 * /sick_safetyscanners/sick_safetyscanners/application_io_data: True
 * /sick_safetyscanners/sick_safetyscanners/channel_enabled: True
 * /sick_safetyscanners/sick_safetyscanners/derived_settings: True
 * /sick_safetyscanners/sick_safetyscanners/frame_id: scan
 * /sick_safetyscanners/sick_safetyscanners/general_system_state: True
 * /sick_safetyscanners/sick_safetyscanners/host_ip: 192.168.1.249
 * /sick_safetyscanners/sick_safetyscanners/host_udp_port: 0
 * /sick_safetyscanners/sick_safetyscanners/intrusion_data: True
 * /sick_safetyscanners/sick_safetyscanners/measurement_data: True
 * /sick_safetyscanners/sick_safetyscanners/sensor_ip: 192.168.1.2
 * /sick_safetyscanners/sick_safetyscanners/skip: 0
 * /sick_safetyscanners/sick_safetyscanners/time_offset: 0.0
 * /sick_safetyscanners/sick_safetyscanners/use_persistent_config: False

NODES
  /sick_safetyscanners/
    sick_safetyscanners (sick_safetyscanners/sick_safetyscanners_node)

ROS_MASTER_URI=http://localhost:11311

process[sick_safetyscanners/sick_safetyscanners-1]: started with pid [16689]
[ WARN] [1569333711.686504342]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO] [1569333711.693093446]: Starting SickSafetyscanners
[ INFO] [1569333711.693381723]: UDP client is setup
[ INFO] [1569333711.693495906]: Started SickSafetyscanners
[ INFO] [1569333711.693619180]: Enter io thread
[ INFO] [1569333711.693654970]: Reading Type code settings
[ INFO] [1569333711.693786369]: TCP client is setup
[ INFO] [1569333711.694407574]: TCP connection successfully established.
[ INFO] [1569333711.728865155]: Successfully opened Cola2 session with sessionID: 2242537567
[ INFO] [1569333711.759295116]: Command Variable Acknowledged.
[ INFO] [1569333711.759447010]: Type Code: MICS3-ABAZ55IZ1
[ INFO] [1569333711.798789829]: Successfully closed Cola2 session with sessionID: 2242537567
[ INFO] [1569333711.798955620]: TCP Connection successfully shutdown
[ INFO] [1569333711.799034118]: TCP Socket successfully closed.
[ INFO] [1569333711.799097295]: TCP client is setup
[ INFO] [1569333711.799847461]: TCP connection successfully established.
[ INFO] [1569333711.849012966]: Successfully opened Cola2 session with sessionID: 1157913675
[ WARN] [1569333711.878848376]: Command Method Not Accepted.
[ INFO] [1569333711.919022374]: Successfully closed Cola2 session with sessionID: 1157913675
[ INFO] [1569333711.919273253]: TCP Connection successfully shutdown
[ INFO] [1569333711.919401363]: TCP Socket successfully closed.
[ INFO] [1569333711.919482751]: Successfully launched node.

puck-fzi, 2019-09-24:
okay, to check that the latest commit did not introduce a regression youl you please test the branch fb_typecode on this repository?

JinTongxing, 2019-09-24:
> okay, to check that the latest commit did not introduce a regression could you please test the branch fb_typecode on this repository?

okay, please wait

JinTongxing, 2019-09-24:
> okay, to check that the latest commit did not introduce a regression could you please test the branch fb_typecode on this repository?

Unfortunately, the problem still exists. =.=

puck-fzi, 2019-09-24:
Okay, then it seems it is no regression in the code. Best would be to switch on the latest commit on the master then, since this is the most up to date code. And then we will try to troubleshoot from there.

Do you have any other components in your system except your scanner and the computer?
Is your connection stable?
Are there any firewalls which might be blocking incoming values?

Other then that have you tried this on a different machine? 
Since I do not have the exact same model of the sensor here i can not check myself if that has something to do with the model. 

What might help is if you could post the TCP response for that TCP query. So that i can have a look into the bytestream

JinTongxing, 2019-09-24:
> Okay, then it seems it is no regression in the code. Best would be to switch on the latest commit on the master then, since this is the most up to date code. And then we will try to troubleshoot [...]

I have tried it on another computer, it still couldn't work. According to one of my colleague's analysis and explanation, the TCP response is wrong. Maybe there is something wrong with my device? (Well it works well through Safety Designer under windows.) My colleague and I would try to deal with this problem, and we could keep in contact. Thanks anyway.

puck-fzi, 2019-09-24:
Yes, the response you are getting will be with an error code, since there is no parsing of the error code implemented for that command, the bytestream of the response would be interesting. to see what the error code is 

Let me know if I can help you further, and if you find any bug or anything which might be not working on the driver side, please let me know so it can be fixed

puck-fzi, 2019-09-24:
Feel free to reopen when you find new insights

closed 2019-09-24 by puck-fzi

JinTongxing, 2019-09-25:
The actual outputs for getCommandType() and getCommandMode() are 'F' and 'A', rather than 'A' and 'I' in catkin_ws/src/sick_safetyscanners-master/src/cola2/MethodCommand.cpp:
`bool MethodCommand::processReply()
{
  bool result = false;
  if ((getCommandType() == 'A' && getCommandMode() == 'I') ||
      (getCommandType() == 0x41 && getCommandMode() == 0x49))
  {
    ROS_INFO("Command Method Acknowledged.");
    result = true;
  }
  else
  {
    ROS_WARN("Command Method Not Accepted.");
  }
  return result;
}`
Maybe it's associated with the problem.

puck-fzi, 2019-09-25:
Could you tell me what the bytes after the command type contain? There should be a number, easiest to check would be in a tcp dump. The F and A basically jsut say that the Command was not accepted

JinTongxing, 2019-09-25:
How could I find them out?

puck-fzi, 2019-09-25:
you could use wireshark or tcpdump. And then look for the tcp message containing the FA send from the sensor to your computer

JinTongxing, 2019-09-25:
The successive two seqs are like:
0000  00 e0 4c 68 01 09 00 06  77 12 15 f3 08 00 45 00   ..Lh.... w.....E.
0010  00 38 27 e6 00 00 40 06  ce 8e c0 a8 01 02 c0 a8   .8'...@. ........
0020  01 f9 08 4a c6 59 0d 4c  79 16 3c bd 5b 66 50 18   ...J.Y.L y.<.[fP.
0030  08 00 95 18 00 00 02 02  02 02 00 00 02 d0 00 00   ........ ........
0040  bc 0b d8 cb 05 83  
and
0000  00 e0 4c 68 01 09 00 06  77 12 15 f3 08 00 45 00   ..Lh.... w.....E.
0010  02 f0 27 e7 00 00 40 06  cb d5 c0 a8 01 02 c0 a8   ..'...@. ........
0020  01 f9 08 4a c6 59 0d 4c  79 26 3c bd 5b 66 50 18   ...J.Y.L y&<.[fP.
0030  08 00 5e cc 00 00 52 41  1b 00 56 01 00 00 05 00   ..^...RA ..V.....
0040  00 00 2d 43 00 00 36 e9  18 03 2d 43 00 00 36 3d   ..-C..6. ..-C..6=
0050  19 03 2d 43 00 00 4d 7c  19 03 00 00 00 00 00 00   ..-C..M| ........
0060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0080  00 00 00 00 00 00 00 00  00 00 5a 80 80 27 00 00   ........ ..Z..'..
0090  00 00 00 00 00 00 00 00  00 00 c4 ea 74 bb 00 00   ........ ....t...
00a0  00 00 00 00 00 00 00 00  00 00 1f 9f d2 d5 6a bb   ........ ......j.
00b0  8e 20 24 b8 47 54 39 6a  f8 85 33 00 6d 00 00 00   . $.GT9j ..3.m...
00c0  00 00 18 4a 03 00 1c 00  00 00 a8 00 00 00 ae 42   ...J.... .......B
00d0  00 00 ab 74 5f 02 ae 42  00 00 95 7c 5f 02 ae 42   ...t_..B ...|_..B
00e0  00 00 7d 6b 60 02 00 00  00 00 03 00 00 00 74 79   ..}k`... ......ty
00f0  74 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   t....... ........
0100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0110  00 00 00 00 00 00 07 be  f1 82 00 00 00 00 00 00   ........ ........
0120  00 00 00 00 00 00 5e 36  e3 e0 00 00 00 00 00 00   ......^6 ........
0130  00 00 00 00 00 00 70 ab  fb 37 9a 98 a7 a5 39 3a   ......p. .7....9:
0140  3a 06 8f 26 57 54 33 01  5b 00 00 00 00 00 25 d7   :..&WT3. [.....%.
0150  0f 00 12 00 00 00 9b 00  00 00 ae 42 00 00 3b ea   ........ ...B..;.
0160  5b 02 ae 42 00 00 60 f2  5b 02 ae 42 00 00 71 1b   [..B..`. [..B..q.
0170  5c 02 00 00 00 00 03 00  00 00 74 79 74 00 00 00   \....... ..tyt...
0180  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0190  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
01a0  00 00 57 c0 ec d4 00 00  00 00 00 00 00 00 00 00   ..W..... ........
01b0  00 00 45 c6 7e 34 00 00  00 00 00 00 00 00 00 00   ..E.~4.. ........
01c0  00 00 da 39 09 32 20 75  ad fa 06 82 71 e0 0a c1   ...9.2 u ....q...
01d0  61 1e 33 01 59 00 00 00  00 00 f1 4c 0c 00 12 00   a.3.Y... ...L....
01e0  00 00 9b 00 00 00 ae 42  00 00 96 1f 53 02 ae 42   .......B ....S..B
01f0  00 00 bb 25 53 02 ae 42  00 00 45 37 59 02 00 00   ...%S..B ..E7Y...
0200  00 00 03 00 00 00 74 79  74 00 00 00 00 00 00 00   ......ty t.......
0210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0220  00 00 00 00 00 00 00 00  00 00 00 00 00 00 f4 da   ........ ........
0230  21 73 00 00 00 00 00 00  00 00 00 00 00 00 47 c8   !s...... ......G.
0240  56 d1 00 00 00 00 00 00  00 00 00 00 00 00 30 78   V....... ......0x
0250  cf 53 4e f7 c1 7a 96 a9  45 be fd d3 4d ff 33 01   .SN..z.. E...M.3.
0260  57 00 00 00 00 00 21 80  03 00 12 00 00 00 9a 00   W.....!. ........
0270  00 00 ae 42 00 00 0b 80  29 02 ae 42 00 00 2f d4   ...B.... )..B../.
0280  29 02 ae 42 00 00 a5 19  2a 02 00 00 00 00 00 00   )..B.... *.......
0290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
02a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
02b0  00 00 00 00 00 00 00 00  00 00 c4 a1 a0 c9 00 00   ........ ........
02c0  00 00 00 00 00 00 00 00  00 00 85 78 8a 28 00 00   ........ ...x.(..
02d0  00 00 00 00 00 00 00 00  00 00 31 1c 38 92 0d cf   ........ ..1.8...
02e0  19 4c 46 8f b7 88 0c 31  5c fe 33 01 4e 00 00 00   .LF....1 \.3.N...
02f0  00 00 f7 1b 27 00 11 00  00 00 9a 00 00 00         ....'... ....
would that help?

JinTongxing, 2019-09-25:
The successive two seqs are like:
0000  00 e0 4c 68 01 09 00 06  77 12 15 f3 08 00 45 00   ..Lh.... w.....E.
0010  00 38 27 e6 00 00 40 06  ce 8e c0 a8 01 02 c0 a8   .8'...@. ........
0020  01 f9 08 4a c6 59 0d 4c  79 16 3c bd 5b 66 50 18   ...J.Y.L y.<.[fP.
0030  08 00 95 18 00 00 02 02  02 02 00 00 02 d0 00 00   ........ ........
0040  bc 0b d8 cb 05 83                                  ......           

and
0000  00 e0 4c 68 01 09 00 06  77 12 15 f3 08 00 45 00   ..Lh.... w.....E.
0010  02 f0 27 e7 00 00 40 06  cb d5 c0 a8 01 02 c0 a8   ..'...@. ........
0020  01 f9 08 4a c6 59 0d 4c  79 26 3c bd 5b 66 50 18   ...J.Y.L y&<.[fP.
0030  08 00 5e cc 00 00 52 41  1b 00 56 01 00 00 05 00   ..^...RA ..V.....
0040  00 00 2d 43 00 00 36 e9  18 03 2d 43 00 00 36 3d   ..-C..6. ..-C..6=
0050  19 03 2d 43 00 00 4d 7c  19 03 00 00 00 00 00 00   ..-C..M| ........
0060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0080  00 00 00 00 00 00 00 00  00 00 5a 80 80 27 00 00   ........ ..Z..'..
0090  00 00 00 00 00 00 00 00  00 00 c4 ea 74 bb 00 00   ........ ....t...
00a0  00 00 00 00 00 00 00 00  00 00 1f 9f d2 d5 6a bb   ........ ......j.
00b0  8e 20 24 b8 47 54 39 6a  f8 85 33 00 6d 00 00 00   . $.GT9j ..3.m...
00c0  00 00 18 4a 03 00 1c 00  00 00 a8 00 00 00 ae 42   ...J.... .......B
00d0  00 00 ab 74 5f 02 ae 42  00 00 95 7c 5f 02 ae 42   ...t_..B ...|_..B
00e0  00 00 7d 6b 60 02 00 00  00 00 03 00 00 00 74 79   ..}k`... ......ty
00f0  74 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   t....... ........
0100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0110  00 00 00 00 00 00 07 be  f1 82 00 00 00 00 00 00   ........ ........
0120  00 00 00 00 00 00 5e 36  e3 e0 00 00 00 00 00 00   ......^6 ........
0130  00 00 00 00 00 00 70 ab  fb 37 9a 98 a7 a5 39 3a   ......p. .7....9:
0140  3a 06 8f 26 57 54 33 01  5b 00 00 00 00 00 25 d7   :..&WT3. [.....%.
0150  0f 00 12 00 00 00 9b 00  00 00 ae 42 00 00 3b ea   ........ ...B..;.
0160  5b 02 ae 42 00 00 60 f2  5b 02 ae 42 00 00 71 1b   [..B..`. [..B..q.
0170  5c 02 00 00 00 00 03 00  00 00 74 79 74 00 00 00   \....... ..tyt...
0180  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0190  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
01a0  00 00 57 c0 ec d4 00 00  00 00 00 00 00 00 00 00   ..W..... ........
01b0  00 00 45 c6 7e 34 00 00  00 00 00 00 00 00 00 00   ..E.~4.. ........
01c0  00 00 da 39 09 32 20 75  ad fa 06 82 71 e0 0a c1   ...9.2 u ....q...
01d0  61 1e 33 01 59 00 00 00  00 00 f1 4c 0c 00 12 00   a.3.Y... ...L....
01e0  00 00 9b 00 00 00 ae 42  00 00 96 1f 53 02 ae 42   .......B ....S..B
01f0  00 00 bb 25 53 02 ae 42  00 00 45 37 59 02 00 00   ...%S..B ..E7Y...
0200  00 00 03 00 00 00 74 79  74 00 00 00 00 00 00 00   ......ty t.......
0210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
0220  00 00 00 00 00 00 00 00  00 00 00 00 00 00 f4 da   ........ ........
0230  21 73 00 00 00 00 00 00  00 00 00 00 00 00 47 c8   !s...... ......G.
0240  56 d1 00 00 00 00 00 00  00 00 00 00 00 00 30 78   V....... ......0x
0250  cf 53 4e f7 c1 7a 96 a9  45 be fd d3 4d ff 33 01   .SN..z.. E...M.3.
0260  57 00 00 00 00 00 21 80  03 00 12 00 00 00 9a 00   W.....!. ........
0270  00 00 ae 42 00 00 0b 80  29 02 ae 42 00 00 2f d4   ...B.... )..B../.
0280  29 02 ae 42 00 00 a5 19  2a 02 00 00 00 00 00 00   )..B.... *.......
0290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
02a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
02b0  00 00 00 00 00 00 00 00  00 00 c4 a1 a0 c9 00 00   ........ ........
02c0  00 00 00 00 00 00 00 00  00 00 85 78 8a 28 00 00   ........ ...x.(..
02d0  00 00 00 00 00 00 00 00  00 00 31 1c 38 92 0d cf   ........ ..1.8...
02e0  19 4c 46 8f b7 88 0c 31  5c fe 33 01 4e 00 00 00   .LF....1 \.3.N...
02f0  00 00 f7 1b 27 00 11 00  00 00 9a 00 00 00         ....'... ....

puck-fzi, 2019-09-25:
Could you maybe provide a screenshot, and I think this is not the correct message, I am looking for a fairly short return, where the beginning of the following sequence is included

[image]

Just the AI should be changed to FA and one of the zeros in the four last bytes should be changed as well

JinTongxing, 2019-09-25:
The log information is just like this by tcpdump, how could I show more information?
[image]

puck-fzi, 2019-09-25:
These messages appear to be just an echo? They all contain the same data?

There should be a message from the sensor to your computer containing the sequence i posted above. best look for the 02 02 02 02 which is the start of the cola2 telegram. 

I mainly use wireshark, but this should be sufficent if you find the correct message from the tcpdump.
And the screenshot already contains messages from which ip the data is send

JinTongxing, 2019-09-25:
Actual there is a little difference between them ...
I also used wireshark under windows, but the lidar works well under windows like in SafetyDesigners, so I think its more meaningful to capture the data under ubuntu by tcpdump...
But it seems that it doesn't have 02 02 02 02, I donnot know why

puck-fzi, 2019-09-25:
This request will only be executed during startup, so you have to capture the rcp sequence when the command methos is invoked, which will be during startup of the driver. Basically i would like to see the respone of the senor for this request. I am not concerned about continous data, it will be a single message being send by the sensor sometime at the beginning

puck-fzi, 2019-09-25:
And only the tcp information is relevant, whatever is passed from the udp data does not matter at this stage

JinTongxing, 2019-09-25:
Here may be something useful, when I restart the driver, i got such TCP log information including 02 02 02 02 
[image]

puck-fzi, 2019-09-25:
yes this looks already better, now there should be a few messages further down. There was the TypeCode Request. The Next request should then be the communication settings

JinTongxing, 2019-09-25:
[image]
[image]
This is the following log information

puck-fzi, 2019-09-25:
Alright I'll try to have a look through the next days

JinTongxing, 2019-09-27:
Is there any solution to deal with this problem?

JinTongxing, 2019-09-27:
As the exception occurs when running 'changeSensorSettings', would that mean that the session don't have the right or authority to change the settings of my lidar?

puck-fzi, 2019-09-27:
I am still looking into it, but i am actually not sure, since i can run the code on my sensor fine.

A small work around for the moment could be, that you configure the sensor how you want it to be with the IPs and ports for the UDP Data and comment out the change sensor setting in the driver. But this means you can not reconfigure or set parameters from the launch file.

Do you have any other devices in your setup?

And that the request is rejected means most likely that the sensor, does not like one of the parameters which are passed to it. But since you do not really change any parameters except the IPs it should work in theory. From the logs i could not really deduct anything yet either

JinTongxing, 2019-09-28:
1. I just comment out 'm_device->changeSensorSettings(m_communication_settings)' in SickSafetyScannerROS, but I still couldn't get any data from topics;
2. I noticed that the UDP port was not fixed to a certain value as TCP does that it could change when I rerun the code? Is that true?
3. Lastly, I wanna know what does the command type F and command mode A in MethodCommand::processReply()  mean?  And where is the cola2 manual that you mentioned in the code? 
Thank you very much!

JinTongxing, 2019-09-29:
Hello, maybe you are on a happy weekend, but I think there should be a fatal problem in the driver. As most clients may just configure the lidar via SafetyDesigner and few people use it under ROS, it may be abnormal to encounter such a case. I  really tried anything I can do including resetting the lidar via SafetyScanner, but no udp packet was received and all the topics are vacuum, I think SICK should really pay attention to the problem especially for this type of device. Thanks for your attention.

puck-fzi, 2019-09-29:
>     1. I just comment out 'm_device->changeSensorSettings(m_communication_settings)' in SickSafetyScannerROS, but I still couldn't get any data from topics. From the code, the aim of changeSensorSetting is to activate continuous output of data via UDP, so maybe it's why no UDP data was sent out.
> 
>     2. I noticed that the UDP port was not fixed to a certain value as TCP does and it could change when I re-run the code? Dose that matter?
>        Thank you very much!

1.) Yes the change sensor settings tries to enable the udp datastream and all necessary outputs needed from the sensor and configures the IPs and Ports. These are all  varaibles which can be set inside the safety designer as well. But to make sure the driver has all data this method is called.

 2.) Yes the UDP Port is taken from the ephemeral range of ports. But it is possible to assign a fixed port to the sensor

puck-fzi, 2019-09-29:
> Hello, maybe you are on a happy weekend, but I think there should be a fatal problem in the driver. As most clients may just configure the lidar via SafetyDesigner and few people use it under ROS, [...]

Well this driver is designed for ROS, and it takes care of the communication between the sensor and ROS. Since i can not reproduce the error on my side and have it working here it is difficult to pinpoint the problem.Since you are the first having these troubles it might be something about your setup etc. If you have any insights, or find an error I am more then happy to try to fix this and therefore improve the driver

JinTongxing, 2019-10-09:
After calling changeSensorSettings(m_communication_settings), I found that LIDAR replies 46 41 01 00 or FA 0001, does that mean the code doesn't have the right authority to call the method as the manual goes? And how could I do to change the user group in ROS driver?

puck-fzi, 2019-10-09:
Hi, the error code refers to "1: An generic error occurred while activating the navigation data con‐
figuration." Sadly the cola manual does not give any more information on that. Only if the return value is 0 the change in the command settings will be executed.

JinTongxing, 2019-10-09:
Ok, and by the way where is the cola manual that you refer to?

puck-fzi, 2019-10-09:
The manual can be found here in the latest version:
https://cdn.sick.com/media/docs/1/01/701/Technical_information_Data_output_via_UDP_and_TCP_IP_en_IM0083701.PDF

JinTongxing, 2019-10-09:
From page 35, I come to the conclusion that 46 41 01 00 means incorrect user group, and where did you get the information  "1: An generic error occurred while activating the navigation data con‐
figuration." ?

puck-fzi, 2019-10-09:
From page 57. there the command change command is written and the return values given.

JinTongxing, 2019-10-09:
> there the command change command is written and the return values given.
All right, thank you!

JinTongxing, 2019-10-09:
As the return information is "46(F) 41(A) 01 00" rather than "41(A) 49(I) b0 00 .. .. .. .. "，is it meaningful to refer to the list in page 57 ?

puck-fzi, 2019-10-09:
Sorry for the confusion. I suppose you are correct. If the method index is not returned it sounds more like the error code. I don't know how to set the user group though. The manual states for all methods the user group run. It might be that this can be configured in the safetydesigner.

Sicne the typecode variable command seemed to work, you have at least read access i suppose.

JinTongxing, 2019-10-09:
Yes, I now have the read access and the right return for type code inquiry.
As you can run the driver successfully, have you done anything more to have the right user group? Or is there any methods to get rid of the user group detection to avoid the error?

thanarnop, 2021-11-12:
I have the same problem when I run command roslaunch manually, It seem to OK and output is appear perfectly, but if I reboot to run by robot upstart on startup  there are only node list and topic name but no output data appear

puck-fzi, 2021-11-12:
Hi, please open a new issue for new requests.

However, in your case it appears that the driver is not at fault if it runs perfectly when you execute roslaunch.
My guess would be that your robot upstart does not wait until all networks are up and the sensor is connected. Therefore, the driver will not connect to the sensor.

thanarnop, 2021-11-12:
Than you