SICKAG/sick_safetyscanners #27: Measurement data with 2 sensors for 2 receivers with UDP broadcast

tbohnma, 2019-12-12:
Hello,

with the current driver I could not acquire scan data from two microScans3 at two receivers. 

Please, refer to the attached pptx for more detailled description. 

[LiDAR-LOC_ROS_test.pptx](https://github.com/SICKAG/sick_safetyscanners/files/3955873/LiDAR-LOC_ROS_test.pptx)

puck-fzi, 2019-12-12:
Hello,

first of all your launch file differs from the one you show in the presentation, which is the one you used for the tests?

Then second, the driver configures one UDP adress in the sensor, to which the UDP Data packages are sent. There is no support for in the driver to send the UDP packages to multiple Hosts. I even think, there is no support in the scanner to send the data to multiple sinks, since the target UDP adress is specified via Cola2.
The way would be to get the raw data on one computer and then send the ros messages to the other nodes on the different machines.

For the UDP Port 0 in both controllers... this allows the driver to select a port  from the ephemeral range, therefore two different ports will be chosen.

Hope this clarifies your questions.

tbohnma, 2019-12-12:
Sorry, for the wrong file. I updated it in the previous post. 

The above launch file, (should) set(s) both sensors to UDP broadcast, which allows multiple receivers of the measurement data. Without ROS, but safety designer it works. So the scanners support this. 

Also it works, with the last procedure, to acquire raw data on both receivers.

Thanks for the info about port=0.

puck-fzi, 2019-12-12:
Alright. i'll have a look into it. If you have find any issue, feel free to open a pull request with your solution.

puck-fzi, 2020-01-30:
Unfortunately I haven't been able to locate the issue until now. I will keep investigating

NikolasE, 2020-02-19:
I'm currently having the same setup (Two NanoScan, one SIM1000 and a ROS Laptop) and problem. I think this is going to a common use case with the ROS-Interfaces for both the NanoScan and the Lidar-Localization (https://github.com/SICKAG/sick_lidar_localization).

I could not test the given solution in the first post as the sick_safetyscanners_org.launch is missing. 

What is happening during the short TCP-connection at the start of this node? I saw in wireshark that my sensor is sending on UDP, so shouldn't it be possible to that this node without the TCP connection and just read the UDP-packages?

puck-fzi, 2020-02-19:
Hi, if you have any insights feel free to share, i am currently trying to solve this issue as well. 
The _org file is the original launch file as provided in this repository.

And yes at the beginning the sensor sends the current configuration as provided by the launch file to the sensor. There it appears something is not written correctly to the sensor and overwrites the persistent sensor config.

A workaround might be to disable the sending of the current sensor config and use the config from the safety designer. But this is in my opinion only a workaround and not the preferred way, since there are more parameters send to the sensor then just the IP addresses.

puck-fzi, 2020-03-02:
Hello together,

i tried to find the issue you are getting. My current setup is a notebook and a tinkerboard both connected via switch to a microScan3 sensor. The sensor is set via SafetyDesigner to Broadcast (192.168.1.255). On the tinkerboard i start then the driver with the following command:

`roslaunch sick_safetyscanners sick_safetyscanners.launch host_ip:="192.168.1.255" host_udp_port:=6070 sensor_ip:="192.168.1.11"`

When I look at wireshark connections I still get all the UDP packages on my notebook. Changed to the corresponding port (6070 in this case).

Therefore I am still not able to reproduce the before mentioned error and the broadcast works in my case. Which might help is to set the host_udp_port to the desired ports, since when it is configured to 0 a port from the ephemeral range will be used. 

Please let me know if that helps. Furthermore if you are starting two instances of the sensor you should do so in separate namespaces with separate ports.

It even was possible to start a ros_driver instance on the tinkerboard and one on the notebook and both gave output on the scan topic. On both computer the before mentioned command was used.

If you still have the same issue please let me know what exactly you tried, in my setup I am not able to reproduce the errors and the broadcast functionality works.

puck-fzi, 2020-03-02:
Additionally, I set the sensor in the SafetyScanner to send datagrams to 192.168.1.100 and then checked wireshark on the notebook (192.168.1.9). 
-> No UDP datagrams

Then launched the driver with on the tinkerboard (192.168.1.100):
`roslaunch sick_safetyscanners sick_safetyscanners.launch host_ip:="192.168.1.255" host_udp_port:=6070 sensor_ip:="192.168.1.11"`

-> UDP datagrams appearing in wireshark on notebook (192.168.1.9)

Therefore even without setting the broadcast in the safety designer the broadcast will be enabled by the driver

yu-yajika, 2020-03-03:
Hello, I'm colleague of author.
Would you please tell me how to disable the sending of the current sensor config? by SOPAS or Telegram?
**_A workaround might be to disable the sending of the current sensor config and use the config from the safety designer._**

And thank you very much for your support, I'll test your procedure.

puck-fzi, 2020-03-03:
Hi, as mentioned it should work with the current state of the driver. If not please let me know.
For the workaround in src/SickSafetyscannersRos.cpp comment out 
`line 90   m_device->changeSensorSettings(m_communication_settings);`

But as mentioned it should not be needed. And it only internal of the ROS driver, that disables the sending of the new settings to the sensor.

puck-fzi, 2020-03-03:
The out of range issue, as mentioned in issue #32 as well, should be fixed with the latest commit. If not please let me know.

puck-fzi, 2020-03-04:
Since the broadcast works i my setup with the current version of the driver, I will close this issue. Feel free to reopen if there are still problems.

closed 2020-03-04 by puck-fzi

DatZ2H, 2020-10-01:
Hello,
I'm currently having the same setup (Four Microscan3, Two SIM1000 and a ROS IPC 18.04LTS) ( This is two AGV) and problem.
The system ONE setup: 
-The First time (Error):
  + MS3 :             IP:              192.168.1.12
                    Subnet:              255.255.255.0
   Data Output:  IP Addrress: 192.168.1.255
                           UDP PORT:  6060
  + MS3 :             IP:              192.168.1.13
                    Subnet:              255.255.255.0
   Data Output:  IP Addrress: 192.168.1.255
                           UDP PORT:  6061
=>    Config in Sick_safetyscanners.launch 
                            Sensor_IP1 : 192.168.1.12
                            Sensor_IP1 : 192.168.1.13
                                   Host_IP: 192.168.1.255
                      Host_UDP_PORT1: 6060
                      Host_UDP_PORT1: 6061
=> Error: 'Std::out_of_range' 
  what():  vector<bool>::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[sick_safetyscanners/sick_safetyscanners-2] process has died 
I fix config in Sick_safetyscanners.launch  is Host_UDP_PORT => i can connect with two Microscan3 but In launch Lidar-Loc, Data is Error. In the same time.
And Then i fix setup: 
 +MS3 (192.168.1.12) => UDP PORT 6070
 +MS3 (192.168.1.13) => UDP PORT 6061
 Launcher :                      Host_UDP_PORT1: 6070
                                        Host_UDP_PORT1: 6061
=> It's OK. Lidar-Loc ok can get position and i can get raw data from Two MS3.
The system TWO setup: 
+ MS3 :             IP:              192.168.1.22
                    Subnet:              255.255.255.0
   Data Output:  IP Addrress: 192.168.1.255
                           UDP PORT:  6070
  + MS3 :             IP:              192.168.1.23
                    Subnet:              255.255.255.0
   Data Output:  IP Addrress: 192.168.1.255
                           UDP PORT:  6061
=>    Config in Sick_safetyscanners.launch 
                            Sensor_IP1 : 192.168.1.22
                            Sensor_IP1 : 192.168.1.23
                                   Host_IP: 192.168.1.255
                      Host_UDP_PORT1: 6070
                      Host_UDP_PORT1: 6061
=> Error: 'Std::out_of_range' 
  what():  vector<bool>::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[sick_safetyscanners/sick_safetyscanners-2] process has died 
I fix config in Sick_safetyscanners.launch  is Host_UDP_PORT => i can connect with two Microscan3 but In launch Lidar-Loc, Data is Error. In the same time.

I tried a few of UDP but It don't work or can connect with MS3 But Lidar-loc Error

Could you check for me.

Thanks

tbohnma, 2020-10-01:
Hello Dat, 
concerning out_of_range have you used the latest driver, since this error was fixed here: https://github.com/SICKAG/sick_safetyscanners/issues/32 

Just for clarification:

- The 2 setups/AGVs are completely independent or are they connected with WiFi?
- 1 system contains:
-- 2x mS3
-- 1x SIM with LiDAR-LOC
-- 1x ROS IPC 18.04LTS
- There are 2 members requesting raw UDP data from the mS3: LiDAR-LOC (for localization) and the ROS IPC (e.g. for visualization in RVIZ)
- System 1 is running without problem, when using ports 6070 and 6061 right?
- System 2 is causing the same problem as in system 1 with config 6060, 6061 and no working UDP ports can be found, right?

DatZ2H, 2020-10-02:
Hi ,
I did it again clone (git clone https://github.com/SICKAG/sick_safetyscanners.git)
But  when i choose UDP is (6060, 6061....) => ERROR #32.
Thist it my system:
https://user-images.githubusercontent.com/52695336/94882696-e5ad2c80-0492-11eb-9c80-62dfa68721cb.png
System 1: running with (6070 vs 6061) but why not is (6060 and 6061) because Sick recommed in Hardware Instruction at the link above in chapter 6.2.4.5 is UDP 6060 and 6061. and When i using 6060 and 6061 with system 1=> Error Out_of_range.
System 2: with only Package Sick_Safetyscanner can connect 2 MS3 with 6060,6061 but Data ong Lidar-Loc Error. I saw in the wireshark on PC it ok. Has Source 192.168.1.22 and 23 destination 192.168.1.255 and info 6060=> and 6061=>6061.

Thanks

puck-fzi, 2020-10-05:
Hi Dat,

Could you please provide the log of the crash, preferably with the driver being build in debug mode.
For this please rebuild the package with 
`catkin_make -DCMAKE_BUILD_TYPE=Debug`

The issues of out of range erros is supposed to be fixed in the latest release. However if issues still persist
I would need the logs to identify the error. Thanks

DatZ2H, 2020-10-07:
Hello,
Thanks for the advise and guidance, Finally I can get Position ( Lidar-Loc) and Raw data from Broadcast.
System1 or System 2 with UDP Port using 6060 and 6061=> Working.
I have done like instruction: https://github.com/SICKAG/sick_safetyscanners/issues/46
In the file: SickSafetyscannerRos.cpp (catkin_ws/src/sick_safetyscanners/src/SickSafetyscannerRos.cpp)
I changed.
OLD: 
if (monitoring_case_number_flags.at(0))
  {
    msg.active_monitoring_case = monitoring_case_numbers.at(0);
  }
NEW:
if (monitoring_case_number_flags.size()>0){
msg.active_monitoring_case = monitoring_case_numbers.at(0);
}else {
msg.active_monitoring_case = 0; 
}
And it can run any UDP

puck-fzi, 2020-10-07:
Hi, great to hear, 
I will try to integrate this check and see why this occurs at this place.
Thanks for your insights.

nguyendo1403, 2023-09-15:
Hi @puck-fzi, when i use Sick Microscan3, can i get Position (Lidar-Loc) and Raw data from Broadcast? Can you show me the instructions? Thank you very much!