SICKAG/sick_safetyscanners #28: NanoScan3 | No Scan Data

christianpfitzner, 2019-12-16:
Hi, 

I have received a test model of the nanoscan3. However, I am not able to get the data from the sensor in ros. 

The issue looks similar as documented here: https://github.com/SICKAG/sick_safetyscanners/issues/23

The scanner is obviously connected correctly, however, no output is generated. 

Listening on the udp port shows that there is somehow a communication. 

Please see the starting of the launch file, including the paramters: 
`SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /sick_safetyscanners/sick_safetyscanners/angle_end: 0.0
 * /sick_safetyscanners/sick_safetyscanners/angle_start: 0.0
 * /sick_safetyscanners/sick_safetyscanners/application_io_data: True
 * /sick_safetyscanners/sick_safetyscanners/channel_enabled: True
 * /sick_safetyscanners/sick_safetyscanners/derived_settings: True
 * /sick_safetyscanners/sick_safetyscanners/frame_id: scan
 * /sick_safetyscanners/sick_safetyscanners/general_system_state: True
 * /sick_safetyscanners/sick_safetyscanners/host_ip: 192.168.1.10
 * /sick_safetyscanners/sick_safetyscanners/host_udp_port: 0
 * /sick_safetyscanners/sick_safetyscanners/intrusion_data: True
 * /sick_safetyscanners/sick_safetyscanners/measurement_data: True
 * /sick_safetyscanners/sick_safetyscanners/sensor_ip: 192.168.1.2
 * /sick_safetyscanners/sick_safetyscanners/skip: 0
 * /sick_safetyscanners/sick_safetyscanners/time_offset: 0.0
 * /sick_safetyscanners/sick_safetyscanners/use_persistent_config: False

NODES
  /sick_safetyscanners/
    sick_safetyscanners (sick_safetyscanners/sick_safetyscanners_node)

ROS_MASTER_URI=http://localhost:11311

process[sick_safetyscanners/sick_safetyscanners-1]: started with pid [1552]
[ WARN] [1576492484.987229749]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO] [1576492484.999352775]: Starting SickSafetyscanners
[ INFO] [1576492484.999494877]: UDP client is setup
[ INFO] [1576492484.999525585]: Started SickSafetyscanners
[ INFO] [1576492484.999604121]: Reading Type code settings
[ INFO] [1576492484.999639004]: Enter io thread
[ INFO] [1576492484.999690161]: TCP client is setup
[ INFO] [1576492485.000357124]: TCP connection successfully established.
[ INFO] [1576492485.001508980]: Successfully opened Cola2 session with sessionID: 2399427849
[ INFO] [1576492485.002035379]: Command Variable Acknowledged.
[ INFO] [1576492485.002837542]: Successfully closed Cola2 session with sessionID: 2399427849
[ INFO] [1576492485.002897177]: TCP Connection successfully shutdown
[ INFO] [1576492485.002930387]: TCP Socket successfully closed.
[ INFO] [1576492485.002954697]: TCP client is setup
[ INFO] [1576492485.003365568]: TCP connection successfully established.
[ INFO] [1576492485.004260937]: Successfully opened Cola2 session with sessionID: 3276363820
[ INFO] [1576492485.005361265]: Command Method Acknowledged.
[ INFO] [1576492485.006459549]: Successfully closed Cola2 session with sessionID: 3276363820
[ INFO] [1576492485.006545748]: TCP Connection successfully shutdown
[ INFO] [1576492485.006589557]: TCP Socket successfully closed.
[ INFO] [1576492485.006609377]: Successfully launched node.
[ INFO] [1576492627.757463752]: TCP client is setup
[ INFO] [1576492627.757842075]: TCP connection successfully established.
[ INFO] [1576492627.759583236]: Successfully opened Cola2 session with sessionID: 2076338344
[ INFO] [1576492627.760358325]: Command Method Acknowledged.
[ INFO] [1576492627.760843857]: Successfully closed Cola2 session with sessionID: 2076338344
[ INFO] [1576492627.760889171]: TCP Connection successfully shutdown
[ INFO] [1576492627.760919604]: TCP Socket successfully closed.
[ INFO] [1576492633.986125313]: TCP client is setup
[ INFO] [1576492633.986562126]: TCP connection successfully established.
[ INFO] [1576492633.987246235]: Successfully opened Cola2 session with sessionID: 347634165
[ INFO] [1576492633.987768173]: Command Method Acknowledged.
[ INFO] [1576492633.988580525]: Successfully closed Cola2 session with sessionID: 347634165
[ INFO] [1576492633.988661921]: TCP Connection successfully shutdown
[ INFO] [1576492633.988707360]: TCP Socket successfully closed.
`

puck-fzi, 2019-12-16:
From the launch of the node, it appears to be started correctly. Compred to the other issue. the methods are all accepted, so it appears to be another issue.

You made sure, that all the IPs are set correctly? Are any firewalls blocking messages?
There was an issue with ufw, see Issue #15

christianpfitzner, 2019-12-16:
Thank's for the quick response. 

I am not running any firewall on my linux system: 
```➜  ~ sudo ufw status 
Status: inactive
```

puck-fzi, 2019-12-16:
Hi, since i can not reproduce the error, and the bringup seems to be correct.

Could you check if the function void `AsyncUDPClient::handleReceive ` in `sick_safetyscanners/src/communication/AsyncUDPClient.cpp` is triggered. This should be the very first function processing the UDP Packets. But it should show a ROS_ERROR when it fails. And for the socket, if it does not come up properly it should throw an exception and show an error. 
To make sure, that the UDP packets actually are passed to the driver.

christianpfitzner, 2019-12-16:
Ok, got it:
I build the package from source and now it works like a charm. 

Thank you very much for your support.

puck-fzi, 2019-12-16:
Thank you for the clarification... seems like there is a needed update then in the source for the nanoscan. I will trigger a new release for the package then.

closed 2019-12-16 by puck-fzi

puck-fzi, 2019-12-16:
The release is triggered, with the next update cycle a version 1.0.4 will be released as a catkin package for kinetic and melodic.