SICKAG/sick_safetyscanners #32: std::out_of_range bug

reinzor, 2020-02-07:
Driver crash:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

puck-fzi, 2020-02-07:
Hi, 
please deliver a more complete issue report. With a simple error code i can not understand what the problem is and where it might occur. From looking at your pull request, i understand where you think the error originates from. Please add a proper stack trace, for example using gdb. Like that it appears that you might made a fix. But since i can not reproduce the error, i can not easily understand where the problem occurs and why your bugfix should solve it.

reinzor, 2020-02-07:
I agree, unfortunately, I do not have the cycles to debug this that thoroughly at the moment.

reinzor, 2020-02-10:
`gdb backtrace`:

```
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.  
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/lowpad/dev/lowpad/devel/lib/sick_safetyscanners/sick_safetyscanners_node...done.
(gdb) run
Starting program: /home/lowpad/dev/lowpad/devel/lib/sick_safetyscanners/sick_safetyscanners_node scan:=front_scan extended_laser_scan:=front_scan/extended_laser_scan output_paths:=front_scan/output_paths 
raw_data:=front_scan/raw_data field_data:=front_scan/field_data __name:=front_scan
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff07a1700 (LWP 28801)]
[New Thread 0x7fffeffa0700 (LWP 28802)]
[New Thread 0x7fffef79f700 (LWP 28803)]
[New Thread 0x7fffeef9e700 (LWP 28804)]
[ WARN][/front_scan][1581341494.315852754]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO][/front_scan][1581341494.323344656]: Starting SickSafetyscanners
[ INFO][/front_scan][1581341494.323686592]: UDP client is setup
[ INFO][/front_scan][1581341494.323799808]: Started SickSafetyscanners
[New Thread 0x7fffee79d700 (LWP 28805)]
[ INFO][/front_scan][1581341494.324042774]: Enter io thread
[ INFO][/front_scan][1581341494.324074425]: Reading Type code settings
[ INFO][/front_scan][1581341494.324182906]: TCP client is setup
[ INFO][/front_scan][1581341494.324593124]: TCP connection successfully established.
[ INFO][/front_scan][1581341494.325696794]: Successfully opened Cola2 session with sessionID: 640621383
[ INFO][/front_scan][1581341494.326862592]: Command Variable Acknowledged.
[ INFO][/front_scan][1581341494.327055259]: Type Code: NANS3-CAAZ30AN1P04
[ INFO][/front_scan][1581341494.327950975]: Successfully closed Cola2 session with sessionID: 640621383
[ INFO][/front_scan][1581341494.328022739]: TCP Connection successfully shutdown
[ INFO][/front_scan][1581341494.328068839]: TCP Socket successfully closed.
[ INFO][/front_scan][1581341494.328110575]: TCP client is setup
[ INFO][/front_scan][1581341494.328468706]: TCP connection successfully established.
[ INFO][/front_scan][1581341494.329279232]: Successfully opened Cola2 session with sessionID: 1524551303
[ INFO][/front_scan][1581341494.330044904]: Command Method Acknowledged.
[ INFO][/front_scan][1581341494.330585082]: Successfully closed Cola2 session with sessionID: 1524551303
[ INFO][/front_scan][1581341494.331233442]: TCP Connection successfully shutdown
[ INFO][/front_scan][1581341494.331287839]: TCP Socket successfully closed.
[ INFO][/front_scan][1581341494.331330048]: Successfully launched node.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
Thread 6 "sick_safetyscan" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffee79d700 (LWP 28805)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff5c33801 in __GI_abort () at abort.c:79 
#2  0x00007ffff6626957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff662cab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff662caf1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff662cd24 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff6628855 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000055555564526c in std::vector<sick::datastructure::ScanPoint, std::allocator<sick::datastructure::ScanPoint> >::_M_range_check (this=0x7fffee79c1a0, __n=18446744073709551615)
    at /usr/include/c++/7/bits/stl_vector.h:825
#8  0x000055555563bc4f in std::vector<sick::datastructure::ScanPoint, std::allocator<sick::datastructure::ScanPoint> >::at (this=0x7fffee79c1a0, __n=18446744073709551615)
    at /usr/include/c++/7/bits/stl_vector.h:846
#9  0x0000555555622a4e in sick::SickSafetyscannersRos::createLaserScanMessage (this=0x7fffffffc870, data=...) at /home/lowpad/dev/lowpad/src/sick_safetyscanners/src/SickSafetyscannersRos.cpp:335
#10 0x0000555555621fc5 in sick::SickSafetyscannersRos::receivedUDPPacket (this=0x7fffffffc870, data=...) at /home/lowpad/dev/lowpad/src/sick_safetyscanners/src/SickSafetyscannersRos.cpp:249
[... 16 lines cut ...]
    at /home/lowpad/dev/lowpad/src/sick_safetyscanners/src/communication/AsyncUDPClient.cpp:81
#23 0x00007ffff7b2d848 in sick::communication::AsyncUDPClient::<lambda(boost::system::error_code, std::size_t)>::operator()(boost::system::error_code, std::size_t) const (__closure=0x7fffee79caf0,
[... cut ...]
#36 0x00007ffff7afeac0 in boost::detail::thread_data<boost::_bi::bind_t<bool, boost::_mfi::mf0<bool, sick::SickSafetyscanners>, boost::_bi::list1<boost::_bi::value<sick::SickSafetyscanners*> > > >::run (
    this=0x5555558e4b00) at /usr/include/boost/thread/detail/thread.hpp:116
#37 0x00007ffff59dfbcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#38 0x00007ffff692a6db in start_thread (arg=0x7fffee79d700) at pthread_create.c:463
#39 0x00007ffff5d1488f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

reinzor, 2020-02-10:
So something appears to be going wrong here: https://github.com/SICKAG/sick_safetyscanners/blob/master/src/SickSafetyscannersRos.cpp#L336

JaccovdS, 2020-02-11:
It looks like that a `unsigned long long  = -1 ` is given as argument to the `at()`. Since that `unsigned long long -1` equals `18446744073709551615`. For a 64-bit system `std::size` is a `unsigned long long`. The question is how the `-1` ends up there in the first place.

puck-fzi, 2020-02-12:
Okay, i found something which might relate to that error
uint16_t num_scan_points = data.getDerivedValuesPtr()->getNumberOfBeams();
uint32_t num_scan_points = data.getMeasurementDataPtr()->getNumberOfBeams();

as defined in the output. the number of beams is once in as uint16_t and once as uint32_t. Since iI havent been able to reproduce the error, could you maybe check what is both of the variables? The fix probably would be to replace the first line with the second. But I am curious what is written in the other number.

puck-fzi, 2020-03-03:
Hi, could you please try again with the latest commit. I changed the code to use ` uint32_t num_scan_points = data.getMeasurementDataPtr()->getNumberOfBeams();` instead of the derived values. Hopefully this solves the problem. 

There should be a ROS warning if the number points of the derived values and the measurement data missmatch. If so please paste the output of the warning. And please state what sensor you are using. Since i never was able to reproduce the error.

reinzor, 2020-03-03:
Thanks. We will put this on the upcoming sprint. Will get back to you within two weeks. We are using a Sick NanoScan 3.

puck-fzi, 2020-03-16:
Hi @reinzor,
did you have any troubles with the updated driver? Is this issue resolved for you?

reinzor, 2020-03-16:
Hi @puck-fzi , we are still testing this. It didn't occur anymore but I would like to give it one more week to make sure that the issue is resolved. I will keep you up to date using this issue.

puck-fzi, 2020-03-16:
Thanks for the feedback. Glad to here it seems to work

reinzor, 2020-03-20:
We have tested this for a week and it did not occur anymore due to your fix. Thanks for your support!

puck-fzi, 2020-03-20:
Happy to hear. If anything else pops up. Let me know.

YannickdeHoop, 2020-05-20:
```
[ WARN][/front_scan][1589963357.400053168]: Points do not match: 1072, 0
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```
We still encounter this issue.

puck-fzi, 2020-05-20:
Hi thanks for reporting, I still not quiet understand why this happens, I will merge a branch into the master, which should solve it since it work with the actual size of the scan_points... and then will have to dig further into it, why this is occuring

puck-fzi, 2020-05-20:
Updated the master accordingly, please try the latest commit 27fa2f87bae940b793055437c5783f2c2bc395cd

Let me know if there are any more errors occurring.

closed 2020-05-20 by puck-fzi

Tiantian-git, 2022-04-14:
Hi @puck-fzi , we still encounter this issue after upgrade our driver to v1.0.8 which is commit [772ba7d]. 
We used the branch v1.0.4 [e2ce149] before, this crash happened almost everyday. After upgrade to v1.0.8, it became better but the driver still crashed around once a week. Could you please help us solve this problem? I have no idea why this happened. Let me know if u need any other information.
Thanks

puck-fzi, 2022-04-14:
Hi Tiantian, please open a new issue for this.

For help, please build the library from source with debug flags enabled and provide a stack trace with gdb to pinpoint where the error is occuring.