SICKAG/sick_safetyscanners #34: Propagate error messages on the display of nanoscan

cssreddyharsha, 2020-02-12:
I was trying to figure out if the ROS driver forwards any error messages that pop up on the display. The general system status block seems to be of size 10 bytes according to the data header of each packet:

`52 2 0 0 1293dd8 12741f1 0 b0f87 b0fb2 0 144fb84 **4c 10** 60 18 7c 1884 1904 124 1a2c 108`.
According to the document for `Data output via UDP and TCP/IP` in microscan, the device status block seems to be much larger and has the error info at 24th byte offset of the general system status block [here.](https://github.com/SICKAG/sick_safetyscanners/blob/e2ce1497bada188804d939e1d547f9ad9ac1a2e6/src/data_processing/ParseGeneralSystemState.cpp#L204) 
The ROS driver also seems to be just setting if there is an application error or a device status error and not reporting it. 
Am I missing something in the document 8022708 or is there a function that reports the error code of the nanoscan?

puck-fzi, 2020-02-12:
According to https://cdn.sick.com/media/docs/1/01/701/Technical_information_Data_output_via_UDP_and_TCP_IP_en_IM0083701.PDF page 21/22, you can see what is written in the general system state. If the error you are looking for is not in one of the fields. It will not be send by the Sensor via UDP packets.

cssreddyharsha, 2020-02-13:
I am interested in the error information reported in the device status message when queried. Is 15th byte of the general system state set to true when any of the errors mentioned on page 45-47 is present on the device. If so I could potentially query whenever this byte is populated. Or I could have a service call that could query this device state. Is there a parser for this message?(I did not see it in cola2).

puck-fzi, 2020-02-13:
The service is not implemented, but the backend provides that functionality.
To get this data a service callback would need to trigger 
```
void SickSafetyscanners::requestStatusOverview(const datastructure::CommSettings& settings,
                                               sick::datastructure::StatusOverview& status_overview)
```
from
`https://github.com/SICKAG/sick_safetyscanners/blob/master/src/SickSafetyscanners.cpp`
This will return you the desired values.
You will find the infos then in this class: `https://github.com/SICKAG/sick_safetyscanners/blob/master/src/datastructure/StatusOverview.cpp`

closed 2020-02-13 by puck-fzi