SICKAG/sick_safetyscanners #35: Time synchronization possible?

jonbinney, 2020-02-14:
It looks like each LaserScan's timestamp is set to the host system time when the data was processed by the driver: https://github.com/SICKAG/sick_safetyscanners/blob/e7054cfd05c492f78ddf1387d3d02561941c9fb7/src/SickSafetyscannersRos.cpp#L306

Two questions:
- does the microscan provide timestamp data for each message it sends?
- is there any way synchronize the microscan's internal clock to the host computer's clock? (NTP or similar?)

I realize the second question isn't related to the driver itself - any pointers would be very helpful. I've got the laser on a moving robot, and would like to synchronize the laser data with our odom data (which is already sync'd to the host cpu) so that we can accurately determine where the laser was when it took each scan.

jonbinney, 2020-02-14:
The datasheet for the profinet version of the microscan mentions getting SNTP added in version 1.2.0 (page 13 [here](https://cdn.sick.com/media/docs/9/79/379/Operating_instructions_microScan3_PROFINET_en_IM0075379.PDF) . But it doesn't say anything else, and It's not clear how to use this feature.

puck-fzi, 2020-02-18:
Yes, currently the time is set in the driver. But you are right, that the microScan sends the timestamp for each scan. You can find an example in a ROS message in https://github.com/SICKAG/sick_safetyscanners/blob/master/msg/DataHeaderMsg.msg
This is used in the RawMicroScanDataMsg, there the information from the complete UDP packet is transfered. But since the sensor is not synchronized I didn't use the sensor time for the LaserScan Message.
Furthermore from the datasheet https://cdn.sick.com/media/docs/1/01/701/Technical_information_Data_output_via_UDP_and_TCP_IP_en_IM0083701.PDF Page 20 it says:

> Time at which the data set was created.The measurements are carried out regularly according to the scancycle time set. There can be a jitter in the time stamp since it isnot synchronized with the beginning of the scan.
Date:
•If real-time clock is available: days since 01.01.1972
•If no real-time clock is available: number of full 24-hour cyclessince device was switched on.
Time:
•If real-time clock is available: milliseconds after midnight.
•If no real-time clock is available: milliseconds since the start ofthe current 24-hour cycle since switching on the device

So your Timestamp will most likely be far off your actual Host Time. So if there is a possibility to synchronize the sensor maybe the people from the SICK Support might be able to help you. If you find a way, feel free to share your knowledge here.

closed 2020-02-18 by puck-fzi

jonbinney, 2020-02-18:
Thanks @puck-fzi - I'll see if I can get any more info from Sick Support. If I'm reading the section of the datasheet you posted correctly, time synchronization between the host and the laser might not help, since the There can be "jitter in the time stamp since it is not synchronized with the beginning of the scan"? 

 @chadrockey have you looked into time synchronization of the lasers at all since we talked last?

chadrockey, 2020-02-18:
I still haven't gotten time to do a laser deepdive.

puck-fzi, 2020-02-18:
Yes, it might not help. I don't know how the time is internally handled by the sensor and when exactly the data is written, again this is more likely a question for the Support team. 
Just by guessing i would expect more of an fixed offset then too much jitter, each scan cycle should take the same amount of time, so the jitter will depend on the write time of the timestamp to the datagram. So depending on your requirements for jitter it could still work.

tulku, 2022-01-12:
An implementation to use the timestamp information in the message was suggested here: https://github.com/SICKAG/sick_safetyscanners/pull/90