SICKAG/sick_safetyscanners #40: Shutting down node: escalating to SIGTERM

Erumeldor, 2020-02-28:
Hi,

when shutting down the node without any sensor connected, it takes some time trying to exit the node. After the short pause the node is killed by SIGTERM.

I am currently not 100% sure if this only happens when no sensor is connected, but at the moment here in my virtual machine I don't have the sensor and it happens every time. Due to the fact that escalating to SIGTERM takes around 15 seconds, that's pretty annoying while setting up an integration workspace. (where I have to restart a lot) :)

It doesn't matter which parameter I pass to the launch file or if I start it directly or with a bigger launchfile. 

(In here, only start sick node, wait some seconds, and stop via CTRL+C)
[image]

I don't know where I could get some more information to provide, but if you have any questions feel free to ask!

puck-fzi, 2020-02-28:
Hi,
may i ask what your use case is, if you are not using any sensor? Is it just for setting up a workspace which is later  then sued with a sensor?
In my experience it did not happen with a sensor being connected. So it might have to be something with the handling of the tcp connection, since this is the error message displayed. So far there was never the focus of the driver being used without a sensor.  The feasible solution for me would be that the node shuts down automatically if no tcp connection can be established. Or even better have a simulated sensor, but this i will not tackle.

Erumeldor, 2020-02-28:
Hi,
Yes, the main purpose at the moment would be to set up the workspace for later use and to test other nodes in the workspace without the sick sensor. Therefore we are not in need of any simulated sensor (because we can use recorded rosbags).
For your option to shut down automatically if no tcp connection can be established: At the moment the node would be in a unrecoverable state anyways, right? Therefore that would probably be the best idea.
Another idea would be to repeatedly look for a connection and try to reconnect, but that's another topic.

No matter which way to go, getting rid of the 15 seconds until SIGTERM would be very nice for the future :)

puck-fzi, 2020-03-03:
Hi there, for your convenience I added a quick fix. This is something which will not be integrated into the master branch and is just for testing it now. To properly solve this issue I would need to refactor some code and this might take some time.

So please be aware as soon as you change to a productive system with a sensor you **must** change to the master branch again.

The quick fix can be found here:
https://github.com/SICKAG/sick_safetyscanners/tree/exit_on_failed_connect

puck-fzi, 2020-05-27:
I will close this for now, since i will not dig deeper into this for now. If you have a working version, feeel free to make a pull request.

closed 2020-05-27 by puck-fzi