SICKAG/sick_safetyscanners #41: Doesn't build until Boost 1.71 (Ubuntu Focal)

mikepurvis, 2020-03-03:
Issue is here:

```
.../src/sick_safetyscanners/src/communication/AsyncTCPClient.cpp:51:91:   required from here
/usr/include/boost/asio/detail/io_object_impl.hpp:87:40: error: const class boost::reference_wrapper<boost::asio::io_context> has no member named get_executor
```

And the same thing in `AsyncUDPClient.cpp`.

puck-fzi, 2020-03-04:
As I understand this change is backwards compatible to earlier boost versions. Tested on my local setup, works fine therefore to assure usage under ubuntu focal with  boost 1.71 it will be merged

closed 2020-03-04 by puck-fzi

mikepurvis, 2020-03-04:
Great, thanks for the quick merges!

JingD, 2020-03-12:
Having the same issue while bringing boost 1570 to 1720, can anyone let me know how this is fixed please? :-)

puck-fzi, 2020-03-12:
It should be merged into the current master.

It is tested with 1.71 from @mikepurvis, I haven't checked 1.72, and only assured backwards compatibility. If there are more changes which need to be made, let me know.

anauta2014, 2020-10-07:
I have code with
ip::tcp::acceptor acc_;
if(acc_.get_io_service().stopped())...
this works with boost 1.69
How can I change this code for boost 1.70+ ?

puck-fzi, 2020-10-07:
Hi anuta, this is appears to not be relevant for this ROS package, is this an independent program of yours?
If so, please refer with your question to the documentation of boost or if that does not help to stackoverflow.

PR #42 (2020-03-04): boost::asio API changes in 1.70+
Fixes #41.