SICKAG/sick_safetyscanners #44: Add information to diagnostics

YannickdeHoop, 2020-04-14:
Hi, 

It would be nice to have some extra information published in the diagnostics. So we can check the serial number of the scanner.
The following data should be sufficient:
https://github.com/SICKAG/sick_safetyscanners/blob/73b40f22354f1fc7a2c5dcff793e817b13a435f6/src/SickSafetyscannersRos.cpp#L411
https://github.com/SICKAG/sick_safetyscanners/blob/73b40f22354f1fc7a2c5dcff793e817b13a435f6/src/SickSafetyscannersRos.cpp#L413

Could you add this or do we have to create a pull request?

puck-fzi, 2020-04-14:
Hi Yannick,

the data you want to have in the diagnostics is only static data? Or should there be any data in there which changes during the run time.
If you are only interested in the serial number you could implement a service which calls the following function. 

https://github.com/SICKAG/sick_safetyscanners/blob/73b40f22354f1fc7a2c5dcff793e817b13a435f6/src/SickSafetyscanners.cpp#L145

So far this is not integrated into the ROS API, but there are functions in the base class to call the different variables using cola2. 
If you think it is more advisable to have that in the diagnostics, i suppose it is meant for ease of use and better understanding which sensor is currently used, please feel free to open a pull request.

Cheers
Lennart

reinzor, 2020-04-14:
> the data you want to have in the diagnostics is only static data?

It is mostly static (serial number etc) but also statuses such as:

https://github.com/SICKAG/sick_safetyscanners/blob/73b40f22354f1fc7a2c5dcff793e817b13a435f6/src/SickSafetyscannersRos.cpp#L478

> If you think it is more advisable to have that in the diagnostics, i suppose it is meant for ease of use and better understanding which sensor is currently used.

We are using the Sick NanoScan3 and we would like to have the serial / other diagnostics information of all components in the `/diagnostics` topic so we can use the default ROS diagnostics tooling. This is also implemented in other device drivers such as the `sick tim driver`, `hokyo urg driver` and `intel realense` devices and is AFAIK more less the ROS standard for device info.

> , please feel free to open a pull request.

Sure. Thanks!

puck-fzi, 2020-05-27:
Thanks for contributing. PR #50  is merged into the master

closed 2020-05-27 by puck-fzi