SICKAG/sick_safetyscanners #46: Weird driver behavior when used with multiple sensor configuration

fzoric8, 2020-05-08:
We've been using `sick_safetyscanners` for quite some time. And we've been using it for 2x MicroScan3 sensor configuration. 
Few days ago, we added NanoScan3 into initial sensor setup, therefore we now use 2xMicroScan3 and 1x NanoScan3. After adding NanoScan3, previously used laser driver was positioned on commit `60089125` e.g. version `1.0.3`, we've encountered already reported and solved `std::out of range` [bug](https://github.com/SICKAG/sick_safetyscanners/issues/32). 

I've updated used drivers to include aformentioned fix on commit `73b40f2`. 

For quite some time, everything worked like charm, 2x microscans and 1x nanoscan worked alright but after turning off PC and turning it on again, driver seemed to crash during start up for 2xMicroscans with same already encountered `std::out_of_range` bug on NanoScan3 as follows: 
```
[ INFO] [1588930906.667625056]: TCP connection successfully established.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector<bool>::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```

This was quite weird as it seems that new updated driver works like charm when used with only one/same type of sensor. For example, it works great when used only with 2x MicroScans3 or 1xNanoScan3. 

I use CI , I've started to go back in history and running docker images/containers that contain old driver (only 2x microscans in configuration on version 1.0.3 commit). With old drivers, everything would start up fine. After destroying such docker container and creating new container with new driver and 2x MicroScan3 and 1xNanoScan3 in configuration everything would start up fine as if previously created TCP connection was somehow reestablished fine and UDP packages have been coming continuously. After shutting down PC and starting it up again, new driver and new sensor configuration result in driver crash for MicroScan3 before successfuly opening Cola2 session. 

To handle that, I've created two different ROS packages, one that contains old driver (1.0.3) for MicroScan3 and one that has new driver for NanoScan3 and when running it up, it seemed to work, until I've shut down PC and started it up again, it resulted with same crash as before on old previously used driver for 2xMicroScan3. **Therefore I suspect there is some quite sneaky issue that's existing when using different sensors in same configuration, as if measurement of another device comes on udp port and results in wrong type which causes driver to crash**. I'm not quite sure what's happening and it's been really frustrating as it seems that for now, only solution is to start up old/new driver with just one sensor type e.g. for 2x MicroScan3 and then after successfully starting that, kill started ROS nodes and start any driver configuration for all sensor types. I should note that all sensors use same network subnet and have predefined UDP ports which have been exposed properly for purposes of docker containerization. 

I'm open to any suggestion which could resolve my issue, and provide us with viable solution that makes usage of `sick_safetyscanners` package easy for multiple different sensor configuration. 
Thank you very much, if anything's not clear in my observation please point out, or if you need more output, logs tell me, because I'm really keen to resolve this as soon as possible. Thank you for your help in advance.

puck-fzi, 2020-05-08:
Hi, i will have a deeper dig into this at some stage. For a quick test, please check out the branch fb_nanoScan3. This might resolve the issue. https://github.com/SICKAG/sick_safetyscanners/tree/fb_nanoScan3

Let me know if it does. If not I will have to go deeper into the problem.

fzoric8, 2020-05-08:
Hi, I've debugged an error and honestly I'm quite confused. It seems that, no matter what, driver falls apart for microScans at this particular code snippet: 
```
if (monitoring_case_number_flags.at(0))
  {
    msg.active_monitoring_case = monitoring_case_numbers.at(0);
  }
``` 
which is located in `createOutputPathsMessage` method. It seems that condition tries to reach unexisting element of a `std::vector`. As It seems, this method is same in old and new and `fb_nanoscan3` is implemented in the same way, therefore it crashes everytime after PC has been turned off and turned on again. It's still quite weird behavior, because it's quite non-deterministic. After running drivers for just one type of sensor, it runs fine, but after including different type of sensor in same launch it falls apart if before that we haven't been able to succesfully start up drivers. 

Solution for now is simple check if `monitoring_case_flage` has elements at particular subindex e.g.
```
if (monitoring_case_number_flags.size()>0){
msg.active_monitoring_case = monitoring_case_numbers.at(0);
}else {
msg.active_monitoring_case = 0; 
}
```
Although this little code intervention fixes the problem I'm still confused what's happening. 
I'm not sure how is `getOutputPathsMessage` related to Sick Safety Designer but maybe it has something to do with `Block Application Data` checkbox in `Data Output field` which has been left blank for now in current application case. I would like to understand safety implications of this change and what is really going on. Thank you for your time. Cheers.

puck-fzi, 2020-05-08:
thanks for reporting, and glad to hear it is working for you now. I will have a deeper look into why this is occurring there. This does not seem like a place where anything like that should happen. But I will have a further look into it.

fzoric8, 2020-05-11:
I'm going to close this issue because after following fix the issue has been resolved, but it's still weird. If you ever catch time to dig deeper in it, please share insights. Thank you very much.

closed 2020-05-11 by fzoric8