SICKAG/sick_safetyscanners #52: cann't reconnect again

comwise, 2020-07-01:
if network occurs some error, the laser can't connect again

[1mprocess[sick_safetyscanners/sick_nanoscan3_s300_front-1]: started with pid [29095]
[1mprocess[sick_safetyscanners/sick_nanoscan3_s300_front-1]: started with pid [29095]
[ WARN] [1593581363.358164280]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.
[ INFO] [1593581363.378536652]: Starting SickSafetyscanners
[ INFO] [1593581363.378904192]: UDP client is setup
[ INFO] [1593581363.379016017]: Started SickSafetyscanners
[ INFO] [1593581363.379188968]: Reading Type code settings
[ INFO] [1593581363.379218841]: Enter io thread
[ INFO] [1593581363.379342146]: TCP client is setup
[ERROR] [1593581366.453157903]: TCP error code: 113
[ERROR] [1593581366.453990421]: Error in tcp handle send and receive: 32

puck-fzi, 2020-07-01:
Hi there,
the tcp error code means: 113 = No route to host
So there appears some network issue, have you tried waiting a short moment before restarting, or tried restarting the sensor.
Otherwise, i assume you already had a connection to the sensor beforehand using the driver? Or does this always occure? Then you should check if you can ping the sensor. And make sure you are in the same network.

comwise, 2020-07-02:
@puck-fzi hi，I think if i disconnected the laser, you should be able to reconnect the device

in my case，The first time I didn't configure the network properly，but when i  configure the network ok， the device doesn't work，This means I have to restart the node to complete the connection， So I don't think it is a good way.

other way,  I look up your code, your connection does not return value judgment, The whole framework does not support disconnection and reconnection , Whether data lose or a network break, 

> void AsyncTCPClient::doConnect()
{
  boost::mutex::scoped_lock lock(m_socket_mutex);
  boost::mutex::scoped_lock lock_connect(m_connect_mutex);
  m_socket_ptr->async_connect(m_remote_endpoint, [this](boost::system::error_code ec) {
    if (ec != boost::system::errc::success)
    {
      ROS_ERROR("TCP error code: %i", ec.value());
    }
    else
    {
      ROS_INFO("TCP connection successfully established.");
    }
    m_connect_condition.notify_all();
  });
  m_connect_condition.wait(lock_connect);
//return value ?  If it is a synchronous network connection, it will be blocked for a long time， how can do it？
}

> void SickSafetyscanners::startTCPConnection(const sick::datastructure::CommSettings& settings)
{
  std::shared_ptr<sick::communication::AsyncTCPClient> async_tcp_client =
    std::make_shared<sick::communication::AsyncTCPClient>(
      boost::bind(&SickSafetyscanners::processTCPPacket, this, _1),
      boost::ref(*m_io_service_ptr),
      settings.getSensorIp(),
      settings.getSensorTcpPort());
  async_tcp_client->doConnect();  //return value ??
  m_session_ptr.reset();
  m_session_ptr = std::make_shared<sick::cola2::Cola2Session>(async_tcp_client);
  m_session_ptr->open();  // if disconnect ，i  open the session which is ok？
}

puck-fzi, 2020-07-02:
Hi, yes in general I agree. Having a method which automatically reconnects would be a nice feature. However this is currently nothing I will pursue. Feel free to make a pull request if you implement this. Currently boost::asio is used to establish the connections. There might be some functionality there which helps to implement this, or you could implement a watchdog which automatically tries to reconnect if it doesn't get the keep alive signal.

closed 2020-07-02 by puck-fzi