SICKAG/sick_safetyscanners #53: Standby mode

reinzor, 2020-07-21:
I found the following in the documentation of the MicroScan3 / NanoScan3 for switching the scanner to standby mode:

[image]

Would you be open for a pull request that exposes a `~set_standby` service interface of type [SetBool](http://docs.ros.org/api/std_srvs/html/srv/SetBool.html)?

We could update the frequency diagnostics bounds by looking at the `m_last_raw_data.general_system_state.standby_mode_active` field and update the `m_expected_frequency` accordingly.

What do you think? Thanks!

puck-fzi, 2020-07-21:
Hi,

first of all could you maybe point me to the documentation where this is stated, since i haven't seen that in the cola2 protocol yet.
Are there other options to set as well? Would be interesting to expose all of them then.

Since you are one of the main contributors to the frequency diagnostics I am happy to integrate a pull request with the updated functionality in the diagnostics.

For the set standby, i think this should be possible, I would like to see the whole documentation to see if there might be more variables to add. However it sounds like a reasonable approach.

reinzor, 2020-07-21:
> first of all could you maybe point me to the documentation where this is stated, since i haven't seen that in the cola2 protocol yet.
> Are there other options to set as well? Would be interesting to expose all of them then.

I only searched for the standby mode:

https://www.sick.com/media/docs/4/74/174/Operating_instructions_microScan3_EtherNet_IP%E2%84%A2_en_IM0075174.PDF (page 180)

puck-fzi, 2020-07-21:
Ah okay, then if I understand it correctly these parameters are only set by the assembly and not via the cola2 protocol. So this is only possible between the control unit (a flexisoft) and the sensor. Please correct me if I understand that wrong. 

The driver only supports the cola2 protocol via tcp. Therefore I don't think it is possible to set this variable by the driver. If you have figured out a way to support this, this might be interesting.

reinzor, 2020-07-21:
Thanks, this clears things up. Then I think we have to stick with toggling the standby mode via IO. I will still send a PR for the frequency diagnostics.

puck-fzi, 2020-07-21:
Alright, thank you.

closed 2020-07-21 by puck-fzi