SICKAG/sick_safetyscanners #57: Support for ROS2

NikolasE, 2020-08-31:
Is there a plan to support ROS2?

eliasdc, 2020-09-01:
You can find a working ROS2 version for eloquent and foxy (see branches) on this fork: https://github.com/Tractonomy/sick_safetyscanners

We haven't created a pull request as we changed quite a lot of code and haven't tested more than one sensor (only nano scan 3). Could be interesting to test multiple sensors and get some feedback.

NikolasE, 2020-09-02:
Hello Elias! Thanks for making your work public! I currently have access to a microScan and could test the foxy branch with it

eliasdc, 2020-09-04:
> Hello Elias! Thanks for making your work public! I currently have access to a microScan and could test the foxy branch with it.
> 
> **Update:** Your package works with a microScan3 : ) [I also tested with a NanoScan3
> 
> [Could you maybe allow issue-tracking on your repository? I had a small problem with missing rosidl-libraries]

enabled issue tracking.

puck-fzi, 2020-09-08:
Hi, big thanks to the current ROS2 fork. For now please use this, there might be a migration later of a ROS2 Version into the SICK repositories

closed 2020-09-08 by puck-fzi

puck-fzi, 2021-01-11:
Hi together,

As of now there is an official ROS2 driver, which can be found here: https://github.com/SICKAG/sick_safetyscanners2
Since there have been some structural changes due to the decision to separate driver from the framework wrapper, the ROS2 driver has been moved to a new repository.
The separation allows for usage as a plain cmake project or together with the ROS2 or ISAAC wrappers.

In the next weeks the driver will be released an apt-package.