SICKAG/sick_safetyscanners #58: terminate called after throwing an instance of 'std::out_of_range'

reinzor, 2020-08-31:
Version: `1.0.5`

```
terminate called after throwing an instance of 'std::out_of_range'
what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

Follow up of #32 

Bug still seems to be present in the latest version

puck-fzi, 2020-09-04:
Hi, am still Not in the Office. Could you maybe Check If the fix_check_measurementdata_length branch does solve this Problem?

If Not i Stack trace would be appriciated. Since i cannot reproduce the Error.

reinzor, 2020-09-04:
No problem. I will try to get a stack trace and/or try the mentioned branch. Problem is really hard to reproduce since it only happens once per week or maybe less. I will keep you updated using this issue.

puck-fzi, 2020-09-14:
Does the branch fix the problem? If yes, i will merge it into the master

reinzor, 2020-09-14:
We have not experienced this issue yet. But it does not occur often so I cannot really validate that the problem is fixed yet.

puck-fzi, 2020-09-14:
Thanks, i'll might ping in a week again, to see if this is resolved. Thanks for letting me know

puck-fzi, 2020-09-21:
I merged the branch into the master. If the issue still persists please reopen

closed 2020-09-21 by puck-fzi

akifh, 2021-05-05:
@puck-fzi It is still present with `1.0.8`

```
May 04 22:28:43: terminate called after throwing an instance of 'std::out_of_range'
May 04 22:28:43:   what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

puck-fzi, 2021-05-05:
Hi could you build the library in debug mode and provide me with a stacktrace. So I can see where the error occurs.
So far I did not experience the error again. Thanks