SICKAG/sick_safetyscanners #59: sick-nano driver  occured segfault

menggtgt, 2020-09-04:
The driver program has an error as shown in the figure, and it died, please help me to solve this problem
[image]

menggtgt, 2020-09-04:
Picture may be blurry.

sick_safetyscan[2639]: segfault at 7fB09B03b001 ip 00007fB0b2c3094e sp 00007f80aa29a390 error 4 in libsick_safetyscanners_core.so[7f80b2a26000+29a000]

puck-fzi, 2020-09-04:
Hi, i am currently not in the Office. Could you maybe Check If the fix_check_measurementdata_length branch does solve this Problem?

If Not a Stacktrace would be appreciated. Since i cannot reproduce the Error. Preferably in a debug build. Thanks

menggtgt, 2020-09-04:
No problem. I will try to get a stack trace and/or try the mentioned branch.  I will keep you updated using this issue. Thanks

menggtgt, 2020-09-07:
Hi, The error appeared again, and the results of debugging with GDB are as follows:

[image]

Thread 6 "sick_safetyscan" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffeaffd700 (LWP 6018)]
0x00007ffff7b3494e in sick::read_write_helper::readUint16LittleEndian (it=0 '\000')
    at /sick_safetyscanners/include/sick_safetyscanners/data_processing/ReadWriteHelper.hpp:276
276	  return (*(it + 1) << 8) + *(it + 0);
(gdb) quit
A debugging session is active.

	Inferior 1 [process 4871] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
================================================================================REQUIRED process [front/sick_safetyscanners-5] has died!
process has finished cleanly
log file: .ros/log/52ddbb94-f0e3-11ea-b809-c400ad2cc288/front-sick_safetyscanners-5*.log
Initiating shutdown!
================================================================================

puck-fzi, 2020-09-07:
Hi, did you use the suggested Branch? And a Stack trace using gdb would be helpful. As it appears there is a read request of a 16bit field which is Out of range. Could you Provide the Stack trace using `backtrace` or `bt` in gdb once the Driver Crashes. Thank you

puck-fzi, 2020-09-08:
possible duplicate of https://github.com/SICKAG/sick_safetyscanners/issues/58

menggtgt, 2020-09-08:
Thank you very much, the suggested branch seems to solve this problem and there is no crash yet. Need to verify a few days

puck-fzi, 2020-09-08:
Glad to hear, please let me know if that fixes it for you in the long term, then i will integrate this branch into the master and trigger a new release.

puck-fzi, 2020-09-14:
Does the branch fix the proplem? If yes i will merge it into the master

menggtgt, 2020-09-14:
yes, the error did not reappear

closed 2020-09-14 by menggtgt

puck-fzi, 2020-09-14:
Glad to Hear. Then i will merge the branch.