SICKAG/sick_safetyscanners #64: Visualizing active fields

reinzor, 2020-11-12:
Hi,

We would like to visualize the active warning / protective fields of the scanners using a `geometry_msgs/PolygonStamped` for example. I've seen you implemented a ServiceServer for requesting the `FieldData`: https://github.com/SICKAG/sick_safetyscanners#advertised-ros-services . But how can we get the active fields? Can this only be done via a service request? If so, would it make sense to implement a ROS publisher that streams the fields that are active so that we can query the field configuration once and then only listen to the active fields for the visualization we would like to publish?

Thanks

puck-fzi, 2020-11-12:
Hi,

i'll answer briefly now, if you need more help, just let me know.

Exactly, to get the geometry of the fields you can call the service. Since the fields cannot be adjusted through ROS it is sufficient to load the geometries from the sensor once. For each field you get the information to which monitoring case it belongs. Sicne it is possible to switch between monitoring cases using the flexisoft etc. Then the `~/output_paths (type: sick_safetyscanners/OutputPathMsg)` gives you insight about the current status of the fields and the active monitoring case. Thereby you can check if a field is breached.

Hope that helps

reinzor, 2020-11-12:
Thanks, this clears things up!

closed 2020-11-12 by reinzor