SICKAG/sick_safetyscanners #66: Add ~/active_field topic to driver

tomdewinter94, 2020-11-13:
As a followup to issue #64

I like to implement two `sensor_msgs/LaserScan` topics: `active_protection_field` and `active_warning_field` to the sick driver. So it would be possible to visualize the active field in RVIZ for example. Is this okay with you guys?

puck-fzi, 2020-11-13:
If you see an additional benefit from this, sure it is possible. 
However, in my opinion it doesn't make sense to publish these in the same frequency as the laserscan message since they should not change that often. So maybe only send the message once the active fields change and then latch the message for late subscriber.
Furthermore, you should think about how to handle different fields which are active in parallel. I think currently 8 fields are supported simultaneously with possibly more in the future. So it should be accounted for that as well, these should be distinguishable.

With this being said, feel free to open a pull request, maybe even right away and label it with WIP. Thereby me and other users can take part in discussing/reviewing the changes made to integrate theses topics.

reinzor, 2020-11-13:
If it is solely for visualization purposes, I think we have two options here:

- Custom rviz plugin in another package (listens to a custom message that contains all data published by the driver)
- Publish a `visualization_msgs/MarkerArray` message (latched)

The latter is I think the easiest to implement; however, not sure whether we would like to introduce the additional dependency on `visualization_msgs`. What do you think?

reinzor, 2020-11-13:
I vote for the first option ;)

puck-fzi, 2020-11-16:
I tend to agree, a standalone rviz plugin sounds like the cleanest option.
Otherwise additional overhead will be generated next to a new dependency.

And the necessary information are already there, they just have to be put into rviz.

tomdewinter94, 2020-11-17:
I agree, especially since we want to use a MarkerArray to visualize the safety fields.

So I've been busy making a RVIZ plugin and found a possible bug in the `~/field_data` service call response. The response contains all configured safety fields as a [FieldMsg](https://github.com/SICKAG/sick_safetyscanners/blob/master/msg/FieldMsg.msg). However, the `start_angle` and `angular_resolution` of all fields seem to be wrong:

```
    start_angle: -1.5707963705062866
    angular_resolution: 7.074021368680405e-07
```
The angular resolution is way to small since the safety fields is expressed in 1650 ranges. Also the field of view of our scanner is 270degrees, so I would expect a start_angle of 0.75pi instead of 0,5pi. 

When I use the start angle and resolution from the `~/laser_scan` topic the fields are visualized the right way.

puck-fzi, 2020-11-17:
Hi, for the error please pull the latest version of the driver. Commit https://github.com/SICKAG/sick_safetyscanners/commit/884f946d3d45a210abc1aabcf9bdef459d723847 fixed an issue with the parsing for the field data, this will be rolled out with the next sync of the ros releases in version 1.0.7.

So if you are using the apt package it will take maybe 2 weeks until the fix is ready, if you are building from source simply pull the latest version. If there is still an error please open a new issue since this is a different topic from the one before. Am closing this issue since a decision seems to be reached.

closed 2020-11-17 by puck-fzi