SICKAG/sick_safetyscanners #67: wrong range with nanoscan3

GraceMelon, 2020-11-19:
Hi,

I'm using nanoscan3 in my project. Wrong range values are returned when using nanoscan3 to measure the range to a highly reflective surface. In my test, a reflective tape(5cm wide) is on one side of a box. Theoretically, all the range values returned by the box surface should be continuous and make a smooth curve. But the following figure is what I get.
[image]
Laser scan points between 800 to 825 are range values returned by the reflective tape. They look quite different from the ranges values return by the box surface.
Any replies or suggestions would be greatly appreciated.
Grace

nickoma90, 2020-11-19:
Grace,

thanks for your comment. This however is not related to the driver but to the device and hence, should be discussed with SICK Support. Could you please forward your request to gbc02.techsupport@sick.de?

Thanks!
Max

puck-fzi, 2020-11-19:
Hi Grace,
as Max has already stated, please contact the tech support of SICK.
Closing this here.

closed 2020-11-19 by puck-fzi