SICKAG/sick_safetyscanners #68: Wrong start_angle in fields off ~/FieldData service call

tomdewinter94, 2021-01-19:
Hallo,

As I mention earlier in issue #66, the `start_angle` values of the `sick_safetyscanners/FieldMsg[] fields` in the `~/FieldData` response, seem to be wrong. This should be fixed in version 1.0.7, however this isn't. Only the resolution bug is fixed in this version. 

Example:
```
fields: 
  - 
    ranges: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.3529999852180481, 0.3529999852180481, 0.3529999852180481, 0.3529999852180481, 0.3529999852180481, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.3540000021457672, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35499998927116394, 0.35600000619888306, 0.35600000619888306, 0.35600000619888306, 0.35600000619888306, 0.35600000619888306, 0.35600000619888306, 0.35600000619888306, 0.35600000619888306, 0.3569999933242798, 0.3569999933242798, 0.3569999933242798, 0.3569999933242798, 0.3569999933242798, 0.3569999933242798, 0.3580000102519989, 0.3580000102519989, 0.3580000102519989, 0.3580000102519989, 0.3580000102519989, 0.3580000102519989, 0.35899999737739563, 0.35899999737739563, 0.35899999737739563, 0.35899999737739563, 0.35899999737739563, 0.36000001430511475, 0.36000001430511475, 0.36000001430511475, 0.36000001430511475, 0.36000001430511475, 0.3610000014305115, 0.3610000014305115, 0.3610000014305115, 0.3610000014305115, 0.3619999885559082, 0.3619999885559082, 0.3619999885559082, 0.3619999885559082, 0.3630000054836273, 0.3630000054836273, 0.3630000054836273, 0.3630000054836273, 0.36399999260902405, 0.36399999260902405, 0.36399999260902405, 0.36399999260902405, 0.36500000953674316, 0.36500000953674316, 0.36500000953674316, 0.36500000953674316, 0.3659999966621399, 0.3659999966621399, 0.3659999966621399, 0.367000013589859, 0.367000013589859, 0.367000013589859, 0.36800000071525574, 0.36800000071525574, 0.36800000071525574, 0.36800000071525574, 0.36899998784065247, 0.36899998784065247, 0.36899998784065247, 0.3700000047683716, 0.3700000047683716, 0.3700000047683716, 0.3709999918937683, 0.3709999918937683, 0.3720000088214874, 0.3720000088214874, 0.3720000088214874, 0.37299999594688416, 0.37299999594688416, 0.37299999594688416, 0.37400001287460327, 0.37400001287460327, 0.37400001287460327, 0.375, 0.375, 0.37599998712539673, 0.37599998712539673, 0.37599998712539673, 0.37700000405311584, 0.37700000405311584, 0.3779999911785126, 0.3779999911785126, 0.3790000081062317, 0.3790000081062317, 0.3790000081062317, 0.3799999952316284, 0.3799999952316284, 0.38100001215934753, 0.38100001215934753, 0.38199999928474426, 0.38199999928474426, 0.382999986410141, 0.382999986410141, 0.382999986410141, 0.3840000033378601, 0.3840000033378601, 0.38499999046325684, 0.38499999046325684, 0.38600000739097595, 0.38600000739097595, 0.3869999945163727, 0.3869999945163727, 0.3880000114440918, 0.3880000114440918, 0.3889999985694885, 0.3889999985694885, 0.38999998569488525, 0.39100000262260437, 0.39100000262260437, 0.3919999897480011, 0.3919999897480011, 0.3930000066757202, 0.3930000066757202, 0.39399999380111694, 0.39399999380111694, 0.39500001072883606, 0.3959999978542328, 0.3959999978542328, 0.3970000147819519, 0.3970000147819519, 0.39800000190734863, 0.39899998903274536, 0.39899998903274536, 0.4000000059604645, 0.4000000059604645, 0.4009999930858612, 0.4020000100135803, 0.4020000100135803, 0.40299999713897705, 0.40400001406669617, 0.40400001406669617, 0.4050000011920929, 0.4059999883174896, 0.4059999883174896, 0.40700000524520874, 0.40799999237060547, 0.40799999237060547, 0.4090000092983246, 0.4099999964237213, 0.4099999964237213, 0.41100001335144043, 0.41200000047683716, 0.4129999876022339, 0.4129999876022339, 0.414000004529953, 0.41499999165534973, 0.41499999165534973, 0.41600000858306885, 0.4169999957084656, 0.4180000126361847, 0.4189999997615814, 0.4189999997615814, 0.41999998688697815, 0.42100000381469727, 0.421999990940094, 0.4230000078678131, 0.4230000078678131, 0.42399999499320984, 0.42500001192092896, 0.4259999990463257, 0.4269999861717224, 0.4269999861717224, 0.42800000309944153, 0.42899999022483826, 0.4300000071525574, 0.4309999942779541, 0.4320000112056732, 0.43299999833106995, 0.4339999854564667, 0.4350000023841858, 0.4350000023841858, 0.4359999895095825, 0.43700000643730164, 0.43799999356269836, 0.4390000104904175, 0.4399999976158142, 0.4410000145435333, 0.44200000166893005, 0.4429999887943268, 0.4440000057220459, 0.4449999928474426, 0.44600000977516174, 0.44699999690055847, 0.4480000138282776, 0.4490000009536743, 0.44999998807907104, 0.45100000500679016, 0.4519999921321869, 0.453000009059906, 0.45399999618530273, 0.45500001311302185, 0.4560000002384186, 0.4569999873638153, 0.45899999141693115, 0.46000000834465027, 0.460999995470047, 0.4620000123977661, 0.46299999952316284, 0.46399998664855957, 0.4650000035762787, 0.46700000762939453, 0.46799999475479126, 0.4690000116825104, 0.4699999988079071, 0.47099998593330383, 0.4729999899864197, 0.4740000069141388, 0.4749999940395355, 0.47600001096725464, 0.4779999852180481, 0.4790000021457672, 0.47999998927116394, 0.48100000619888306, 0.4830000102519989, 0.48399999737739563, 0.48500001430511475, 0.4869999885559082, 0.4880000054836273, 0.48899999260902405, 0.4909999966621399, 0.492000013589859, 0.49399998784065247, 0.4950000047683716, 0.4959999918937683, 0.49799999594688416, 0.49900001287460327, 0.5009999871253967, 0.5019999742507935, 0.5040000081062317, 0.5049999952316284, 0.5070000290870667, 0.5080000162124634, 0.5099999904632568, 0.5109999775886536, 0.5130000114440918, 0.5139999985694885, 0.515999972820282, 0.5180000066757202, 0.5189999938011169, 0.5210000276565552, 0.5230000019073486, 0.5239999890327454, 0.5260000228881836, 0.527999997138977, 0.5289999842643738, 0.531000018119812, 0.5329999923706055, 0.5339999794960022, 0.5360000133514404, 0.5379999876022339, 0.5400000214576721, 0.5419999957084656, 0.5440000295639038, 0.5450000166893005, 0.546999990940094, 0.5490000247955322, 0.5509999990463257, 0.5529999732971191, 0.5550000071525574, 0.5569999814033508, 0.5590000152587891, 0.5609999895095825, 0.5630000233650208, 0.5649999976158142, 0.5669999718666077, 0.5690000057220459, 0.5709999799728394, 0.5730000138282776, 0.5759999752044678, 0.578000009059906, 0.5799999833106995, 0.5820000171661377, 0.5839999914169312, 0.5870000123977661, 0.5889999866485596, 0.5910000205039978, 0.5929999947547913, 0.5960000157356262, 0.5979999899864197, 0.6010000109672546, 0.6029999852180481, 0.6050000190734863, 0.6079999804496765, 0.6100000143051147, 0.6140000224113464, 0.6129999756813049, 0.6119999885559082, 0.6100000143051147, 0.609000027179718, 0.6079999804496765, 0.6069999933242798, 0.6060000061988831, 0.6039999723434448, 0.6029999852180481, 0.6019999980926514, 0.6010000109672546, 0.6000000238418579, 0.5989999771118164, 0.597000002861023, 0.5960000157356262, 0 [... cut ...] 0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    start_angle: -1.5620704889297485
    angular_resolution: 0.0029088794253766537
    protective_field: False
```

I'm using the NanoScan3, which has a view of 270degrees. So the start_angle should be about -0.75*pi = -2.35. Right?

puck-fzi, 2021-01-19:
Yes you are right, the field of view is 270 degrees. Just tested it locally on a MicroScan3, there I get the start angle correct with 
```
    start_angle: -2.399827718734741
    angular_resolution: 0.0017453276086598635
```
A related issue might be: https://github.com/SICKAG/sick_safetyscanners2/issues/2
I will have a further look into it. Maybe you can state which versions you are using especially in regards to:
* ros driver
* firmware version
* nanoScan3 Type

puck-fzi, 2021-01-21:
Hi, since i can't reproduce the error I am still not entirely sure what is happening. I removed a few conversions of data types.
Please check out https://github.com/SICKAG/sick_safetyscanners/tree/fb_field_data_start_angle and let me know if that resolves anything.

tomdewinter94, 2021-01-21:
Hello,

Sorry for my late response. 

Here's the scanner info you've asked for:
```
Version version | R
Version major version | 2
Version minor version | 0
Version release | 0
Serial number of device | 19500093
Serial number of channel plug | 19460044
ROS driver = 1.0.7
```
I've also checked our driver parameters:
```
sensor_ip: 10.39.46.16
host_ip: 10.39.46.1
host_udp_port: 6061
frame_id: front_scan
angle_start: -1.55334
angle_end: 1.55334 # angle_start and angle_end == 0 --> all angels are regarded
expected_frequency: 34.043974
time_offset: -0.02
```

As you can see we configure the `start_angle = -1.55334`, because the scanner in our vehicle is limited in view. However, the `~/FieldData` returns the field configuration, which is always the full range. Can you check if you can reproduce the problem if you configure the same range? Probably it's related, but it should not be...

puck-fzi, 2021-01-21:
Okay, yes it seems that I had a misunderstanding on how the values are coupled. I though they are dependent on the current configuration. I confirmed with SICK that they always range the whole field of view of the scanner. Therefore I made them constant.

Please test on confirm the latest commit on the feature branch https://github.com/SICKAG/sick_safetyscanners/tree/fb_field_data_start_angle. If this is as you would expect it to be, I will merge this into the master and it will be further integrated into the base cpp driver.

tomdewinter94, 2021-01-28:
Hello,

It took some time to be able to test it. But the fix seem to be working, thanks!

puck-fzi, 2021-01-28:
Merged into master branch

closed 2021-01-28 by puck-fzi