SICKAG/sick_safetyscanners #69: No map received

arunkumarbak, 2021-01-23:
Hi there,
I am trying create a map using sick safety scanner i am able to subscribe to the sick_safetyscanners/scan topic but when I tried to subscribe to /map topic I am receiving the message as warning 
**No map received**
I want to create map using this scanner.
Is there any changes I want to do in the launch file or in any other files?

thanks and regards,
Arunkumar. B
[image]

puck-fzi, 2021-01-23:
The topics provides by this driver can be seen here: https://github.com/SICKAG/sick_safetyscanners#advertised-ros-topics
The driver itself provides no map, for this you need to apply a SLAM algorithm to your software stack.
If you have troubles with the SLAM have a look at https://answers.ros.org/questions/ if your question was already answered and if not, post a new one. or have a look at your used SLAM implementations github.

closed 2021-01-23 by puck-fzi

arunkumarbak, 2021-01-23:
Hey thanks for the reply,

I am already have my slam algorithm. 
I tried  a few things in my SLAM algorithm but all failed.
I need to know how to create a map using /map topic with this driver .
For that hereby I am attaching my mapping.launch file

`<launch>
    
    <node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">
        <param name="base_frame" value="/base_footprint" />
        <param name="odom_frame" value="/odom" />
        <param name="map_update_interval" value="15.0"/>
        <param name="maxUrange" value="5.0"/>
        <param name="minRange" value="-0.5"/>
        <param name="sigma" value="0.05"/>
        <param name="kernelSize" value="1"/>
        <param name="lstep" value="0.05"/>
        <param name="astep" value="0.05"/>
        <param name="iterations" value="5"/>
        <param name="lsigma" value="0.075"/>
        <param name="ogain" value="3.0"/>
        <param name="lskip" value="0"/>
        <param name="minimumScore" value="100"/>
        <param name="srr" value="0.01"/>
        <param name="srt" value="0.02"/>
        <param name="str" value="0.01"/>
        <param name="stt" value="0.02"/>
        <param name="linearUpdate" value="0.7"/>
        <param name="angularUpdate" value="0.7"/>
        <param name="temporalUpdate" value="-0.5"/>
        <param name="resampleThreshold" value="0.5"/>
        <param name="particles" value="50"/>
        <param name="xmin" value="-50.0"/>
        <param name="ymin" value="-50.0"/>
        <param name="xmax" value="50.0"/>
        <param name="ymax" value="50.0"/>
        <param name="delta" value="0.05"/>
        <param name="llsamplerange" value="0.05"/>
        <param name="llsamplestep" value="0.05"/>
        <param name="lasamplerange" value="0.005"/>
        <param name="lasamplestep" value="0.005"/>
        <param name="transform_publish_period" value="0.1"/>
    </node>
</launch>`

thanks and regards,
Arunkumar B

puck-fzi, 2021-01-23:
Yes, however this is not the place for the mapping questions. Please refere to answers.ros.org or the github of gmapping.
The issues here are for the driver of the safetyscanners itself. Since you are getting the scan topic the driver seems to be working fine.

arunkumarbak, 2021-01-23:
Okay than you for your reply.
So the driver created won't give me the map data required for mapping??
then please took this as my feedback ,
Kindly update the driver for mapping data also.

Thanks and regards,
Arunkumar B

puck-fzi, 2021-01-23:
No the driver itself provides the scan topic. This is of the sensor_msgs/LaserScan type which is the common interface most SLAM algorithm will subscribe to.

> Kindly update the driver for mapping data also.

No this is not the right scope for a driver. To stay in the modular spirit of ROS the driver providing the scan topics should be independent of the mapping algorithms. Thereby you can either exchange the used laser scanners or the used mapping algorithm

arunkumarbak, 2021-01-23:
okay thank you FYI .

Just tell me one thing.
Is it possible to create map using /map topic  using this driver by means of my own algorithms like modifying the launch file or something like that??

Thanks and Regards,
Arunkumar B

puck-fzi, 2021-01-23:
You can adjust the launch files to include gmapping etc. Though I would advise to use a top level launch file including all necessary other launch files. Have a look at the ros tutorials, they migth answer your questions: http://wiki.ros.org/ROS/Tutorials/Roslaunch%20tips%20for%20larger%20projects, or have a look at the navigation tutorial: http://wiki.ros.org/navigation/Tutorials

But in any case you will need both the driver and a mapping algorithm in your ROS stack.