SICKAG/sick_safetyscanners #70: Sick safetyscanner ros driver timestamp

Tiantian-git, 2021-01-28:
Hi there,
I am using sick nanoscan3 with ros driver.
Someone in SICK told me the timestamp for each output from ros driver is set at beam 0 which is around -47.5 degree. But I looked into the code,it looks like the timestamp is set after all of the laser points in one circle are received, which means the timestamp should be around the last beam. 
Could you please help me figure out which is corrected? 
Thank you so much!

Scripts in SickSafetyscannersRos.cpp:
[image]

puck-fzi, 2021-01-28:
Hi, the timestamp is set on message creation, this will be after the all beams are received.

As added in this pull request (https://github.com/SICKAG/sick_safetyscanners/pull/2) you can adjust the time stamp to account for latencie by setting `m_time_offset`
> Many sensors have system dependent but roughly constant time latencies for publication data. When a robot is moving and especially rotating, the timestamp delay from network latency, CPU overhead, or processing delay can cause phase shifts in the laser data and cause blurring/smearing when visualized or used for mapping. This parameter allows users to set this on their system according to their needs. Usually is set by dragging the slider in dynamic reconfigure until an approximate time latency is discovered.

puck-fzi, 2021-02-02:
If there are additional questions please reopen this. Otherwise i deem it as solved

closed 2021-02-02 by puck-fzi