SICKAG/sick_safetyscanners #72: SICK SAFESCAN3 LIDAR using serial communication

JasDEng, 2021-01-30:
Hello!

I am currently using Sick SafeScan3 LIDAR, which is of s300 standard. 
It is said that it uses serial port for communication which means it does not have an IP address (I am not sure if I am right). 
I've connected it and read the user's guide but still has no clue on how to launch it correctly with sick_safetyscanners driver on ubuntu system as it always requires an IP address of the LIDAR and the host.
Would you please help with the issue? I really appreciate it.

regards,
Jasmine

puck-fzi, 2021-01-30:
Hi,
The s300 is not supported by this Driver. This Driver supports the microScan3, nanoScan3 and outdoorScan3.

I think the correct driver for the s300 is https://github.com/ipa320/cob_driver/tree/kinetic_dev/cob_sick_s300

closed 2021-01-30 by puck-fzi

JasDEng, 2021-01-30:
Thank you very much! That really help me a lot!