SICKAG/sick_safetyscanners #74: Multiple receivers: 2 data output channels with nanoScan

tbohnma, 2021-02-02:
Hello, 

the nanoScan offers 2 channels to provide measurement data to 2 targets. 
[image]

How can I configure this in the ROS driver? I only see the option to enable a channel with boolean, but which on will be adressed?

- https://github.com/SICKAG/sick_safetyscanners

[image]

puck-fzi, 2021-02-02:
Hi, 
please try to set the `channel ` parameter to the respective channel [0,1].
Since this was functionality which was done initially but never tested since the microScan3 had no multiple channels at develop time this is not entirely tested. The channels internally will start with the index 0 on the ROS side, which is the default first channel. So setting it to 1 should change the setting. Please let me know if that works.
If yes I will integrate it into the documentation and in the dynamic reconfigure.

puck-fzi, 2021-02-15:
Did this resolve your issue? If yes I will do the integration into the documentation

tbohnma, 2021-02-15:
My colleague is not in office currently. If you have a sensor on hand, feel free to test it yourself. Otherwise I will come back to you the following weeks after we tested this internally.

puck-fzi, 2021-02-15:
Alright thanks, I just have a MicroScan3 at hand, which i think doesn't support the functionality. I will check that. 

Leaving this open and will ping you in a few weeks to see if this can be resolved.

puck-fzi, 2021-03-15:
Setting the channel parameter will enable to configure the respective channel from the driver side.

I will integrate this into the documentation and adapt the launch files accordingly.

Further tests are still pending

puck-fzi, 2021-03-30:
Added the channel flag to the launch file.

Multiple receivers are working on different systems. Tested with two independent ROS systems on different machines. Both connecting to a separate channel with different angle configurations. Working for both machines

closed 2021-03-30 by puck-fzi

nguyendo1403, 2023-09-14:
hi @tbohnma, i'm trying to change monitoring case of sick microscan3, I run package sick_safetyscanner with ROS. Can you help me change the monitoring case, please? Thank you very much.

tbohnma, 2023-10-04:
Hello @nguyendo1403, this ticket is closed. 
It seems you raised this question in several issues. Please use the already provided answer in  https://github.com/SICKAG/sick_safetyscanners/issues/137

Thanks!