SICKAG/sick_safetyscanners #77: Understanding intensities in ROS messages

yashmanian, 2021-02-05:
I'm trying to wrap my head around a few things regarding the intensities sent out by the SICK safety ROS node as a part of the message, and I have a couple of questions:
1. As a part of the `/sick_safetyscanners/scan` message, I see the array of `float32` meant for intensities being published. However, I wasn't able to find documentation on what the saturation value (max intensity) is. It would be helpful if someone was able to tell me what this value is.
2. Looking into the `/sick_safetyscanners/extended_laser_scan` message, I am not quite sure what the `reflektor_status`, `reflektor_median` and the `intrusion` arrays are supposed to indicate. Where can I find some documentation on these?

puck-fzi, 2021-02-09:
Hi,

1.) I am not entirely sure. I think it might range between 0 and 255.0. Will test it the next days what kind of values I am getting, currently not close to a scanner. Otherwise you could ask SICK directly, since the values are the reflectivity the scanner sends.

2.) The extended message contains values for different measuring outputs:
reflector status: Indicates if a scan point is a reflector, this is directly send by the scanner. Which means that the scanner detected a high reflectivity.
reflector_median: returns the middle scan point inside an reflector area.
intrusion: This return if a beam violates one of the protective or warning fields.

Hope this helps.

puck-fzi, 2021-02-15:
Closing this as it seems resolved, if it didn't clarify your questions please reopen.

closed 2021-02-15 by puck-fzi