SICKAG/sick_safetyscanners #79: ROS node does not recover from power-cycle

kalectro, 2021-03-01:
When disconnecting the power to the scanners and reconnecting it, the ROS node is getting stuck. I would expect it to recover or fail when the scanner loses power.

Steps to reproduce:

1. launch the sick_safetyscanners ROS node
2. rostopic hz /scan (working)
3. power-off the scanner -> rostopic hz will show no new messages
4. power-on the scanner -> rostopic hz will continue showing no new messages
5. rosnode kill
6. rostopic hz will show messages again

puck-fzi, 2021-03-02:
Hi,
thanks for opening the issue. Yes there is no watchdog implemented currently. 
For implementing this there would be a few edge cases to be checked. 

The communication from the sensor is based on UDP, therefore there is no check if the connection itself drops.
There are a few options to implement this: 
* Check if UDP packages arrive. However, there are dynamic reconfigure options to disable output from the scanner or sending the packages to different IP addresses. In this case the node should not stop and try to recover. 
* Periodically send a COLA2/TCP request to check if the sensor is alive.
* Use system tool to check if a ping or so can be reached.

Is the sensor configuration from the safetydesigner the same as you are setting from ROS side, especially the IP addresses and which channels are enabled. And further are you using a defined UDP port and that is in the sensors persistent configuration as well. Then it might actually work through a power cycle.

puck-fzi, 2021-03-16:
Since I am currently not planning to implement this, I am closing this.

If you plan on submitting a MR I am happy to discuss more details.

closed 2021-03-16 by puck-fzi

dbojadzic, 2022-09-16:
Is there a way to circumvent this by using the ROS2 package and lifecycle nodes to restart the communication?

puck-fzi, 2022-09-16:
Hi, this is not implemented, you would need to implement the checks for reconnections. It is neither in the normal Nodes nor in the lifecycle nodes implemented yet