SICKAG/sick_safetyscanners #80: TCP error code: 111

JasDEng, 2021-03-27:
Hello!

I am currently using sick_safetyscanners driver to launch a S300 that is connected with a ZLAN module (so it has an IP). I came across a problem, would you please help? I really appreciate it.

Its IP has been set to 192.168.0.3, and I can ping it with command ping 192.168.0.3
But I tried launch the sick_safetyscanners driver, it always has an error, which is shown below. I am not sure what the code means and cannot solve it.

[image]

Thank you!

Best wishes,
Jasmine Deng

puck-fzi, 2021-03-27:
Hi, the s300 is not supported by this driver. This driver is for the microScan3, nanoScan3 and outdoorScan3 safetyscanners. For the s300 have a look at this driver: https://github.com/ipa320/cob_driver/tree/kinetic_dev/cob_sick_s300

closed 2021-03-27 by puck-fzi

JasDEng, 2021-03-28:
Hello, 
Thank you very much for replying!
Did you mean that, after connected to ZLAN module and give it an IP, I should still use cob_sick_s300 rigint?
Thank you!
Best wishes,
Jasmine

puck-fzi, 2021-03-28:
Hi, you best have to ask the people from that driver this. I don't know much about the details.
I just can tell you that this driver does not support the s300 since it implements a different protocol. The microScan3 are communicating via UDP datagrams which is as far as i know only the case for these new safetyscanners.

JasDEng, 2021-03-29:
Thank you very much for replying quickly!
Your reply really helped me a lot.

Best wishes,
Jamine