SICKAG/sick_safetyscanners #81: M2- Data packets of data output lost

BigBbaymax, 2021-04-01:
Dear all,
   I get issue with nano 3.I don't know how to solve the problem.

puck-fzi, 2021-04-01:
Hi, this can happen. However this is a scanner related Issue. Please refer to the  SICK subsidiary where you bought the scanner for this.

closed 2021-04-01 by puck-fzi

yaley386, 2021-04-02:
How to disable ros parameter? use the local output parameter.
[image]

yaley386, 2021-04-02:
ros driver can change the local output parameter.

puck-fzi, 2021-04-03:
I don't quiet understand your question. 
You can set different parameters from the launch file. There you can use the `use_persistent_config` flag to use the defined angles from the Safety Designer.