SICKAG/sick_safetyscanners #82: ros2 and ros driver

Raymond-601-cmd, 2021-04-15:
Dear Colleague,
Could you tell me the difference between the 2 kinds of drivers:
_1. ROS2 driver for SICK safety laser scanners
2. ROS driver for SICK safety laser scanners_
Thanks.

puck-fzi, 2021-04-15:
Hi,

1. ROS2 driver for SICK safety laser scanners
This driver targets the ROS2 Releases starting from Foxy Fitzroy

2. ROS driver for SICK safety laser scanners
This driver targets the ROS1 Releases Kinetic, Melodic and Noetic

closed 2021-04-15 by puck-fzi

Raymond-601-cmd, 2021-04-15:
Dear,
Foxy Fitzroy, Kinetic, Melodic and Noetic
These means?

puck-fzi, 2021-04-15:
These drivers are based around the open source communication framework ROS (https://www.ros.org/). And there they target different releases. If you are instead looking at a standalone Driver for the safetyscanners have a look at the CPP driver: https://github.com/SICKAG/sick_safetyscanners_base

Raymond-601-cmd, 2021-04-15:
Got it. Thanks.