SICKAG/sick_safetyscanners #84: ros driver stop

Raymond-601-cmd, 2021-07-08:
Dear,
We are using SICK nanoScan3 core IO on our AGVs, and we just use the ros driver from github for navigation system.
But the ros driver stopped suddenly durign the running of AGV recently as below picture:
[image]
[image]
[image]
 Could you help to analyze this issue?
Thanks.

puck-fzi, 2021-07-08:
Hi there,

from the provides screenshot I can not tell what happened. Except that there was a crash with a SIGABRT. So most likely some internal signal triggered the abort through an assert or something similar. However, I can not deduct from the provided info why this was happening.

Is this a reoccurring issue? Or did this happen only once so far?
If this happens frequently please make sure you have the latest version of the driver and for debugging purposes best build it from source with debug flags enabled. Then you can get a trace with gdb, which might help to analyze the issue.

Raymond-601-cmd, 2021-07-08:
> Hi there,
>
> from the provides screenshot I can not tell what happened. Except that there was a crash with a SIGABRT. So most likely some internal signal triggered the abort through an assert or [...]

Thank you for your support.
Yes, It happened several times but not regular.  You mean we need to tranck the log for this issue,right?

puck-fzi, 2021-08-09:
Sorry for the late reply.

Yes you need to have debug flags enabled and provide a stack trace. Otherwise it will be impossible for me to pinpoint where the error resides.

so best build the driver from source using `CMAKE_BUILD_TYPE=Debug`

and start the driver with `gdb`

puck-fzi, 2021-09-06:
Closing this, if there are further problems please reopen

closed 2021-09-06 by puck-fzi