SICKAG/sick_safetyscanners #85: Can`t run two sick_safetyscanners at the same time

fatgenius, 2021-08-17:
hi,
I do have two sick safety lidar in my robot. however I can`t start this two lidar with one launch file at same time. i do have two lidar launch file  those two different sick node was start with no problem. but sometimes there`s one lidar no data received. can someone help me with this issue.

puck-fzi, 2021-08-17:
Hi there,

yes you need a separate launch file for each lidar. each node can only process one lidar. 
For the no data received, did you check that all IPs are set correctly and that the ports are different for both sensors?

fatgenius, 2021-08-17:
> Hi there,
> 
> yes you need a separate launch file for each lidar. each node can only process one lidar.
> For the no data received, did you check that all IPs are set correctly and that the ports are different for both sensors?

Hi there,
Thanks in advance. I do have separate launch file with different ip address. when I separate launch it works good.  but when i put those two launch file in one bringup it`s dosen`t work well . Due to occasionally there`s one lidar no data received.

puck-fzi, 2021-08-17:
Are the two nodes in separate namespaces then? this should help. If you post the contents of your launch file i can have a look through it

puck-fzi, 2021-08-18:
On the first look it should be working. Don't see any apparent issues.
Below is a setup of a main launch file I used for an application, maybe that helps. Mainly it just sets two node in two namespaces.
And for the omitted parameters the default values will be used

```
<launch>
  <node pkg="sick_safetyscanners" type="sick_safetyscanners_node" name="sick_safetyscanners" output="screen" ns="sick_safetyscanners_front">
     <param name="sensor_ip" type="string" value="10.42.42.11" />
     <param name="host_ip" type="string" value="10.42.42.9" />
     <param name="host_udp_port" type="int" value="0" />
  </node>

  <node pkg="sick_safetyscanners" type="sick_safetyscanners_node" name="sick_safetyscanners_2" output="screen" ns="sick_safetyscanners_back">
     <param name="sensor_ip" type="string" value="10.42.42.12" />
     <param name="host_ip" type="string" value="10.42.42.9" />
     <param name="host_udp_port" type="int" value="0" />
  </node>

  <!-- node name="rqt_reconfigure" pkg="rqt_reconfigure" type="rqt_reconfigure" /-->
</launch>
```

fatgenius, 2021-08-19:
> On the first look it should be working. Don't see any apparent issues.
> Below is a setup of a main launch file I used for an application, maybe that helps. Mainly it just sets two node in two namespaces.
> And for the omitted parameters the default values will be used
> 
> ```
> <launch>
>   <node pkg="sick_safetyscanners" type="sick_safetyscanners_node" name="sick_safetyscanners" output="screen" ns="sick_safetyscanners_front">
>      <param name="sensor_ip" type="string" value="10.42.42.11" />
>      <param name="host_ip" type="string" value="10.42.42.9" />
>      <param name="host_udp_port" type="int" value="0" />
>   </node>
> 
>   <node pkg="sick_safetyscanners" type="sick_safetyscanners_node" name="sick_safetyscanners_2" output="screen" ns="sick_safetyscanners_back">
>      <param name="sensor_ip" type="string" value="10.42.42.12" />
>      <param name="host_ip" type="string" value="10.42.42.9" />
>      <param name="host_udp_port" type="int" value="0" />
>   </node>
> 
>   
> </launch>
> ```

Thanks again, it`s helps a lot. so far it`s working much more better than better. Since this problem is happens occasionally, I will do a long time test and let`s you know how it works.

reinzor, 2021-08-19:
FYI, we are running two of these nodes on a robot in production and never encountered the problems you mention.

puck-fzi, 2021-09-06:
This should not be stuck there, is your connection stable to the sensor? can you ping it throughout the process?

I will have a look into the code if I can find anything, however I was so far not able to reproduce this behaviour

GitACDreamer, 2021-09-17:
@puck-fzi I encountered the same problem, and it still can’t reproduce。I tried according to your above method, but the problem is still not solved。A sensor has no data, it appears once every few days, irregularly.
@fatgenius Is your problem solved now?

fatgenius, 2021-09-17:
hi，the problem didn’t solve yet. But getting better now. Before it happens
multiple times one day. Nowadays is once few days. I’m trying to change my
router. I’ll keep updating the test results of my  new router.

GitACDreamer, 2021-09-18:
> hi，the problem didn’t solve yet. But getting better now. Before it happens multiple times one day. Nowadays is once few days. I’m trying to change my router. I’ll keep updating the test results of my new router. 

tkx, anyway.

puck-fzi, 2021-12-22:
Hi, did you manage to find any insights? Please reopen if new insights to the problem are made

closed 2021-12-22 by puck-fzi