SICKAG/sick_safetyscanners #86: channel 2 data output lost

Raymond-601-cmd, 2021-08-31:
Dear，
We are using the nanoScan3 from SICK with our AGV controller, but sometimes the controller can not get the data output of channel 2(port 6061) of nanoScan3, the data output of channel 1(port 6060) is always ok.
our navigation driver for nanoScan3 was written by ourselves based on the C++ library of nanoScan3.

Could you help to figure out which aspects can cause this issue?
thanks.

puck-fzi, 2021-09-06:
Hi, for a detailed look i would need some more details. Are there any outputs on the command line from the scanner which appear to be faulty? can you detect, that the packages arrive at the computer through the network interface when on the second channel? 
Is the channel enabled?

puck-fzi, 2021-12-22:
Closing this due inactivity

closed 2021-12-22 by puck-fzi