SICKAG/sick_safetyscanners #92: NanoScan3 no output data when startup on reboot

thanarnop, 2021-11-12:
I have the problem when I manually run command roslaunch the output data is appears my system run perfectly no problem, but when let the robot auto run on startup by the robot upstart start system ,It has no output data appear but the rosnode  and the topic is appear like no problem seem like  running has complete but no output data 
[image]

puck-fzi, 2021-11-12:
Hi,

this looks like an issue with your upstart settings. My best guess is that your startup does not wait for all network interfaces to be ready. So that the driver does not find a connection to the sensor on boot-up. If its a systemd service, try to change the after target to something which indicates that the network is ready.

thanarnop, 2021-11-12:
OK I get it , I will try so

closed 2021-12-22 by puck-fzi

kaorusha, 2022-01-05:
Hi, I had the same problem as you. I tried to modify `/lib/systemd/system/lidar.service` as below but can't solve this problem. Could you share how you solve that?
```sh
[Unit]
Description="bringup lidar"
After=network.target
After=remote-fs.target # added from the original robot_upstart generated file
After=syslog.target # added from the original robot_upstart generated file

[Service]
Type=simple
ExecStart=/usr/sbin/ldlidar-start

[Install]
WantedBy=multi-user.target
```