SICKAG/sick_safetyscanners #97: Missing data when using field of view settings on microScan3

mehditlili, 2022-04-08:
Hi, we are using SICK microScan3 (MICS3-CBAZ55ZA1P01) with this repository and have noticed that when trying to set the field of view angles, a big chunk of data disappears (the latest 45 degrees of the field of view).

When I set both angle_start and angle_end to equal values (disabling field of view filtering) I see all the points. 

But the moment I set angle start and angle_end to -2.39 and 2.39, a lot of points disappear (approximately the last 45 degrees of the fov so between 1.95 and 2.39).

Apart from that, the field of view function itself works as intended and I see points gradually filtered as I increase the angle_start value or decrease angle_end value. 

I looked at the code but it wasn't clear to me where the actual filtering of the points is being done. Can you give me some hints?
thanks

puck-fzi, 2022-06-20:
Hi is this, still present on your part? I could not reproduce the mentioned error.
Is everything shown correctly in the safety designer view?

puck-fzi, 2022-07-19:
Closing due to inactivity