SICKAG/sick_safetyscanners #99: time_increment issue

yifeilang, 2022-05-27:
Hi,
We are seeing a discrepancy between `header.time_increment` and our own calculated result.
From `scan.header`, we could know the scan time of each frame is 0.02999 sec, and we have 1651 scan points each frame. 
Therefore, the time increment between adjacent points should be `0.02999 / 1651 = 1.8165e-5`. 
We got a different time_increment value from scan header, `1.4000e-5`.
Could you help me figure out what cause the difference?
[image]

puck-fzi, 2022-05-30:
Hi,
since both the time_increment and the scan_time are directly taken from the sensor output I am not sure where the error resides. There is no additional handling of the times except transforming them from microseconds to seconds.

puck-fzi, 2022-07-19:
Closing, feel free to reopen if new insights are gained

closed 2022-07-19 by puck-fzi