SICKAG/sick_safetyscanners #101: Issues with Port UDP when use LLS RTE only IPC

MDBean, 2022-07-16:
Hi,
My hardware configure:

MS3 : IP: 192.168.1.12
Subnet: 255.255.255.0
Data Output: IP Addrress: 192.168.1.255
UDP PORT: 6060

MS3 : IP: 192.168.1.13
Subnet: 255.255.255.0
Data Output: IP Addrress: 192.168.1.255
UDP PORT: 6061

IPC: 192.168.1.100
Subnet: 255.255.255.0
IPC OS Ubuntu 18.04 (installed  lidarloc_2.2.0.12R_linux_x86_64.zip) 

And run package "sick_safetyscanners.launch" with "sensor_ip = 192.168.1.12", "host_ip=192.168.1.255", "host_udp_port=6060"
If  run sick_safetyscanners.launch first then run SopasAir => SopasAir cannot connect to scanner MS3_Front.
If  run SopasAir first then run sick_safetyscanners.launch => cannot run sick_safetyscanners.launch.
Because  sick_safetyscanners.launch and SopasAir use the same UDP port (6060).

Before I using LLS with SIM: using sending port 6060 that configurate in MS3, with system using SIM there are 02 device: SIM + IPC but now with LLS RTE only IPC: using same port 6060 as running LLS 2.0 or sick_safetyscanners, cannot run both together as I receive data from port 6060 then split data into 2 different ports but doing this will be delayed.

So how to run  sick_safetyscanners.launch, SopasAir, sick_lidar_localization.launch at same time in only IPC.
I found the information about "optional for further applications e.g: classification, collision avoidance,.." in the standard manual  "Operating Instructions" chapter 4. And how to use them.
[image]

puck-fzi, 2022-07-19:
Hi, I'm not entirely sure what is happening in your system.

As you already figured, broadcasting and multicasting is supported by the sensor, maybe there are some connections, hubs, computers not properly configured to accept multicast?

For running the everything in one system I am not entirely sure this is easily done. Since the driver opens a port on 6060. maybe you have to add options to the port, that other sockets can listen to the same UDP port. So maybe you need to adapt code of the driver since this might not be supported directly in one machine

MDBean, 2022-07-20:
Thank you for the information.

closed 2022-08-22 by puck-fzi