SICKAG/sick_safetyscanners #102: sick nanoscan3 LIDAR stops working after a period of time.

kayoussef, 2022-07-19:
Hi!

We are using nanoScan3 and encounter the situations where the node "sick_safetyscanners" would stop after couple of days of correct work. The moment it happens is hard to capture, so I currently have no log or stack trace to show. Did similar issue appear before?
Note: "respawn" attribute of roslaunch did not make the node reappear, killing and restarting the launch desn't help either. Only a complete power shutdown and restart of the system helped.

puck-fzi, 2022-07-19:
Hi, such as issue is unknown to me, the only issues I have heard of were fixable with a restart of the ROS driver. 
Without logs its hard to tell what might have happened there. 

What are you rebooting, the sensor? Or the computer where the driver is running?

puck-fzi, 2022-10-04:
Closing due to inactivity, if you have more information, please reopen

closed 2022-10-04 by puck-fzi